bigtreetech/MicroProbe #2: Manual and step file dimension deviation

ldehaas1612, 2023-01-17:
The dimensions specified in the manual file and the dimensions of the provided step file are not comparable. Please specify which document to follow for designing a mount.

Also, please provide extra instruction on placement of probe relative to nozzle (minimum/maximum Z height from nozzle, minimum distance away from heatsource, etc.)

Thanks

BADcustomcreations, 2023-08-06:
Manual states the probe should be 1.5 to 2mm below nozzle when pin is fully extended, and the extends 4mm
SO the probe pin when retracted should be 2mm about nozzle

BADcustomcreations, 2023-08-06:
> The dimensions specified in the manual file and the dimensions of the provided step file are not comparable. Please specify which document to follow for designing a mount.
>
> Also, please provide [...]

Manual states the probe should be 1.5 to 2mm below nozzle when pin is fully extended, and the extends 4mm
SO the probe pin when retracted should be 2mm about nozzle