bigtreetech/MicroProbe #3: New install on SKR3 won't home Z

Theo9909, 2023-03-26:
I just finished the install on my printer with an SKR3 board. I followed the directions to a T and have triple checked the firmware. Z axis only goes up and when it goes down, it stops after about 15 mm then says probing failed. Everything is wired properly.

Theo9909, 2023-03-31:
Got it figured out after making about 10 changes to the firmware with the new Marlin Bugfix. Hope this helps others that are stuck trying to get it to work with the new bugfix (3/29/31) version. Below are the additional changes that are not in the manual. Follow the manual to finish the setup of the probe. 
Comment out ENDSTOPPULLUPS
Uncomment USE_PROBE_FOR_Z_HOMING
Do not change the Z_MIN_PROBE_ENDSTOP_HIT_STATE to low even though the BTT manual tells you to. Switching it to High allowed it to work.

#define USE_PROBE_FOR_Z_HOMING
#define Z_MIN_PROBE_ENDSTOP_HIT_STATE HIGH
//#define ENDSTOPPULLUPS

vovodroid, 2023-09-13:
> Switching it to High allowed it to work.

Which version do you have? I have latest v2, and it has Low state and should me connected to z-stop, not to probe pin.

[image]

Theo9909, 2023-09-22:
I have version 1 of the microprobe. I was using the marlin bugfix and that also broke some dual Z motors behavior one night. Got it all figured out then switched to klipper. Much easier for me to make changes and get it working.