bigtreetech/MicroProbe #4: Cant get Microprobe V2 to work on SKR Mini E3 v2

Jimbolaya88, 2023-06-25:
Hello Im familiar with compiling firmware but im not a pro.

I have successfully gotten Microprobe v1 to work and it worked great for months but I broke the pin. I bought a new microprobe thinking i could just replace the whole thing, but after reading the new manual (for V2) there is new firmware updates needed, most notably changing the 
#define Z_MIN_PROBE_ENDSTOP_INVERTING (From False -> True) for V2.
I have ensured my firmware matches all the updates stated in the manual.

The probe will extend, move Z up, then move Z down (slowly) and result in fail because it does not touch the bed within a certain distance.

What I have noticed is if I physically pull out the pin prior to homing, it moves down (fast) searching for the bed untill it is triggered, and after touching, moves up/down slowly to verify bed location. So the my conclusion is the probing sequence is missing the initial deploy to search for bed. Can someone help?

norston, 2023-08-29:
I too am having the same issue. After you send the home Z command the axis rises and then moves back down towards the print bed (i3 type printer like an ender3/prusa mk3). The probe extends just before the axis starts moving back down. I feel that the probe needs to extend sooner, or the gantry needs to not move upwards. And, just like Jimbolaya88, if you pull the pin out manually it works just fine.

Hman32, 2023-09-16:
@Jimbolaya88 did you checked playing with:
  #define Z_MIN_PROBE_ENDSTOP_HIT_STATE HIGH

mrnoobit, 2024-01-20:
Any resolution on this? I’m having the same problem. z_min_probe_endstate_hit_state high/low does nothing for me. Matched everything per biqu.

ViktorDomzoro01081981, 2024-03-17:
give your firmware!

cesko12321, 2024-03-28:
I have the same problem with skr mini e3 v3.0 and i cant compile marlin because i never got into that how can someone help i can even pay 5 10 dollars if you can compile me firmware that works

norston, 2024-03-28:
I used the z min endstop plug instead of the probe port for signaling the z
min.

alkhimey, 2024-04-23:
I have a similar issue with SKR Mini E3 V3.0 and V2.0 Microprobe on Ender 3.

https://github.com/bigtreetech/MicroProbe/assets/652603/8dded8b7-54c9-46e9-91af-0b1ade8b98f4

Musicman3200, 2024-10-29:
Has anyone been able to answer the questions from this forum?  I've hit the exact same issues and have tried MANY different groups of configuration and marlin files to no benefit.  Sorry this is cross posted on another one of these forum postings, anyone had any final luck?

Jimbolaya88, 2024-10-29:
I think this link is finally what worked for me. But this was a while back.
https://www.reddit.com/r/BIGTREETECH/s/KgGRenbY0a