bigtreetech/MicroProbe #9: The Pin Stuck

fiveorders, 2023-08-13:
Marlin Firmware,

Everything was set properly; it did the homing successfully, did the bed levelling successfully. Out of a sudden, the PIN started to get stuck inside when homing. When I pull the pin manually with a tweezer, it works and does the homing, but when the pin is stowed, it does not deploy the pin and can't home the Z axis, it thinks that the pin triggered because the pin is stowed.

It is not Firmware related because it worked properly before this. Can it be a faulty item?

Gadjet5, 2023-09-22:
I'm having the same issue as well was working fine for a out a week and now it gets stuck (Klipper: Endstop z still triggered after retract).
considering putting a little oil on the pin...

Gadjet5, 2023-09-22:
After taking the microprobe off and having a look. it seems that the pcb is attached to the housing via a few dabs of glue(6? maybe. My pcb is loose in the housing so the probe scrapes against its exit hole. So in a pinch, get it realigned and apply a little glue again to hold the pcb inplace. But still looks like a quality control/design fault...

midicdj1000, 2024-04-15:
HI, I have the same problem, the pin is very hard to move and is no longer working. everything was ok yesterday, i switched on and noticed it didn't click up - down at the start today.

tome9111991, 2024-06-18:
I have same issue.
But my pin moves soomth.

After deploy and stow 3x - 4x manuel  via klipper probe works nice for few hours.

I use microprobe in voron 2 .4
Issue almost after a long print with heated chamber

SelfDestructor, 2024-06-26:
This just happened on my second micro probe. It also still moves smoothly and retracts fine, but doesn't deploy. Sometimes it will pop out after a minute or two on it's own or if I tap it.

zeno-tek, 2025-04-23:
I have now had two fail like this, right after the other.