bigtreetech/MicroProbe #11: Microprobe dont run on SKR mini E3 V3 under Klipper

digitalBulldog, 2023-09-06:
hey guys,

here is my entry for the microprobe in Klipper, but its not stopping the homing when touching the bed.
the pin is doesnt works and my toolhead crash into bed.

do you have anyone fix this problem?

#####################################################################
#   Z Stepper Settings
#####################################################################

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
#endstop_pin: PC2
#position_endstop: 0.0
position_max: 150
position_min: 0
homing_speed: 10
second_homing_speed: 3.0
homing_retract_dist: 3.0

[tmc2209 stepper_z]
uart_pin: PC11
tx_pin: PC10
uart_address: 1
run_current: 0.580
stealthchop_threshold: 999999      

#####################################################################
#   BLtouch                                                         #
#####################################################################

[bltouch]
sensor_pin: PC14
control_pin: !PA1
x_offset: -44
y_offset: -7
#Higher the offset, the closer to the bed the nozzle will be.
#Lower the offset, the further the nozzle will be.
#If you need more squish or if it's too high, increase the offset.
#If it's too close, lower the offset.
#z_offset: 1.13
#z_offset: 1.62
pin_move_time: 0.4
probe_with_touch_mode: True
pin_up_touch_mode_reports_triggered: False
stow_on_each_sample: False

please help me, im crying....in need my printer to printing

digitalBulldog, 2023-09-06:
If I use this setting, the same thing happens

## --- BIQU Microprobe Klipper configuration ---
# Not much different from a standard probe config. The
# main difference compared to most probes is to disable
# deactivating the probe (PROBE_UP) after each sample

# You can either copy pasta this directly or import
# into your config.yml using [include microprobe.cfg]

# Note: if you are changing from a bltouch or one of its
# clones, remove/comment out the [bltouch] config!
# That also includes any auto generated config stored
# at the bottom by the PROBE_CALIBRATE command

[gcode_macro PROBE_DOWN]
gcode:
    SET_PIN PIN=probe_enable VALUE=1

[gcode_macro PROBE_UP]
gcode:
    SET_PIN PIN=probe_enable VALUE=0

[output_pin probe_enable]
pin: PA1  # Set to the control pin on your board
value: 0

[probe]
pin: PC14  # Change this to the sensor pin for your board
deactivate_on_each_sample: False
x_offset: -43
y_offset: -7
z_offset: 0  # Will be overridden when you do a PROBE_CALIBRATE
samples: 2
samples_tolerance: 0.05
samples_tolerance_retries: 3
activate_gcode:
    PROBE_DOWN
    G4 P500
deactivate_gcode:
    PROBE_UP

digitalBulldog, 2023-09-07:
[image]
[image]

which microprobe is this now? 
why doesnt answer anyone from biqu or other users?
i'm really confused with this shit! sorry

digitalBulldog, 2023-09-07:
[image]

it is a V2, but i need help with Klipper, please help me anyone...please....i need my printer, its important

vovodroid, 2023-09-13:
I'm not familiar with Clipper, but in Marlin contrary to Biqu manual this new v2 has triggered state Low, and probe is connected to Z-stop, not probe pin.

KLENY199511, 2023-12-25:
mine works like a bee on this config:

[gcode_macro PROBE_DOWN]
gcode:
    SET_PIN PIN=probe_enable VALUE=1

[gcode_macro PROBE_UP]
gcode:
    SET_PIN PIN=probe_enable VALUE=0

[output_pin probe_enable]
pin: PA1  # Set to the control pin on your board
value: 0

[probe]
pin: PC14  # NOTE FOR V2 users: Set this to ^!PC14 to set the low level trigger 
deactivate_on_each_sample: false
x_offset: -39
y_offset: -14
#z_offset: 0  # Will be overridden when you do a PROBE_CALIBRATE
samples: 3
samples_tolerance: 0.01
samples_tolerance_retries: 3
activate_gcode:
    PROBE_DOWN
    G4 P500
deactivate_gcode:
    PROBE_UP

[safe_z_home]
home_xy_position: 150,120 # Change coordinates to the center of your print bed
speed: 50
z_hop: 10                 # Move up 10mm
z_hop_speed: 10
move_to_previous: False