bigtreetech/MicroProbe #13: BIQU B1 SKR V1.4 WITH MICROPROBE V2 FIRMWARE NEEDED! please!

fabianovm, 2023-09-16:
Hello Biqu team, I have a B1, just received the microprobe V2 and it is very difficult for me (and a looot of people) in order to edit the firmware to accept the microprobe V2 firmware modification needs. If you kindly release the final firmware, the .bin for the B1 skr1.4+microprobe V2 would be AWESOME. count on you on that, best regards, thanks!

CS7AFM, 2023-09-22:
I would like also the final firmware so I can enable the microprobe v2 I'va just aquired from biqu store

soulchamp, 2023-09-29:
have either of you had any luck yet on your own. i thought my board was a different version and when i went to install my microprobe tonight i realized i'm in the same boat as you

CS7AFM, 2023-09-29:
> have either of you had any luck yet on your own. i thought my board was a different version and when i went to install my microprobe tonight i realized i'm in the same boat as you

Not yet, I'm very disapointed that the manufacturer have not informed of this complexity and have not resolved it. If this manufacturer is about only seeling and not supporting it's something to think about

fabianovm, 2023-09-30:
Hi, I am really disappointed with this... I spent tons of hours, made a lot of custom firmwares, I was thinking that it was better than BL touch but the cost is high. Have opened the printer, tried to connect probe to z-endstop as some people says, but after days, and days , I gave up. It almost worked for me as a probe, using custom firmware, some homing issues, but I cannot accept that the TFT colorful screen does not work at all to control the microprobe. For me it is unaceptable, how can BIQU sell a microprobe for the BIQU B1, same manufacture. Just want money, if they worry about customers, at least a firmware ready to flash for their own printer B1. I cannot believe in this.

soulchamp, 2023-09-30:
I’ve already decided to return mine and have a BL touch on the way. It’s sad that they have plug and play readiness for the BL touch on the B1 but not their own probe.

fabianovm, 2023-09-30:
If I could, I also would return it. But I am in brazil and ordered in aliexpress
I will wait months, years until this probe is really supported. Wish all luck for you guys.
Regarding the readiness, it is crazy, if you look this repository , you will find a firmware for ender3 and v2 microprobe ...lol... but not for their own brand printer.

CrickYeti, 2023-11-25:
well it is the end of November has anyone had any luck with being able to download the firmware?

PalosX, 2023-11-25:
Hell I can't believe there's not one damn you-tube video for the probe and Marlin set-up. I can't get mine to work in Z homing starts going done then kills printer. Should of known Bigtree would drop the ball with support, hey you get what you pay for right? They always put stuff out to market way to soon

John-Vinic, 2023-12-02:
I need Firmware for SKR 1.4  Biqu helpppppppppppppppppppppppppppppp

fabianovm, 2023-12-02:
And nothing regarding firmware after months.... and also.... I almost made it work, when I remembered that the TFT TOUCHSCREEN also needs to be written because it just works with bl touch.  Gave up, it is a Shame.

knotart01, 2023-12-14:
yo tengo una tarjeta skr 1.4 turbo y la micro probe V2. la versión de marlín 2.1.2.

en https://github.com/section endstops pullups utilizo esta configuración

// Enable pullup for all endstops to prevent a floating state
//#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
// Disable ENDSTOPPULLUPS to set pullups individually
#define ENDSTOPPULLUP_XMIN
#define ENDSTOPPULLUP_YMIN
#define ENDSTOPPULLUP_ZMIN
//#define ENDSTOPPULLUP_IMIN
//#define ENDSTOPPULLUP_JMIN
//#define ENDSTOPPULLUP_KMIN
//#define ENDSTOPPULLUP_UMIN
//#define ENDSTOPPULLUP_VMIN
//#define ENDSTOPPULLUP_WMIN
//#define ENDSTOPPULLUP_XMAX
//#define ENDSTOPPULLUP_YMAX
//#define ENDSTOPPULLUP_ZMAX
//#define ENDSTOPPULLUP_IMAX
//#define ENDSTOPPULLUP_JMAX
//#define ENDSTOPPULLUP_KMAX
//#define ENDSTOPPULLUP_UMAX
//#define ENDSTOPPULLUP_VMAX
//#define ENDSTOPPULLUP_WMAX
#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

..................................

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define I_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define J_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define K_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define U_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define V_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define W_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define I_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define J_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define K_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define U_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define V_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define W_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe.

en Z Probe Options utilizo esto

//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING
Z_MIN_PROBE_PIN

..........................................................................................

Define this pin if the probe is not connected to Z_MIN_PIN.
If not defined the default pin for the selected MOTHERBOARD
will be used. Most of the time the default is what you want.
The simplest option is to use a free endstop connector.
Use 5V for powered (usually inductive) sensors.
RAMPS 1.3/1.4 boards may use the 5V, GND, and Aux4->D32 pin:
For simple switches connect...
 - normally-closed switches to GND and D32.
 - normally-open switches to 5V and D32.
*/
#define Z_MIN_PROBE_PIN P0_10 // Pin 32 is the RAMPS defaul
..............................................................................................................
#define FIX_MOUNTED_PROBE

............................................................................
/**

Probe Enable / Disable
The probe only provides a triggered signal when enabled.
*/
#define PROBE_ENABLE_DISABLE
#if ENABLED(PROBE_ENABLE_DISABLE)
#define PROBE_ENABLE_PIN P2_00 // Override the default pin here
#endif
conecte la impresora 3d a pronter face y ejecute el comando M119, si la sonda esta retraída el estado del z_probe será disparado. si aplica el comando M401 y la sonda desplegara la punta, vuelva a ejecutar el M119 y vera el estado de z_probe abierto. si esto le funciona ejecute el g28 debería funcionar la sonda. saludos

John-Vinic, 2023-12-14:
> yo tengo una tarjeta skr 1.4 turbo y la micro probe V2. la versión de marlín 2.1.2.
> 
> en section endstops pullups utilizo esta configuración
> 
> // Enable pullup for all endstops to prevent a floating state //#define ENDSTOPPULLUPS #if DISABLED(ENDSTOPPULLUPS) // Disable ENDSTOPPULLUPS to set pullups individually #define ENDSTOPPULLUP_XMIN #define ENDSTOPPULLUP_YMIN #define ENDSTOPPULLUP_ZMIN //#define ENDSTOPPULLUP_IMIN //#define ENDSTOPPULLUP_JMIN //#define ENDSTOPPULLUP_KMIN //#define ENDSTOPPULLUP_UMIN //#define ENDSTOPPULLUP_VMIN //#define ENDSTOPPULLUP_WMIN //#define ENDSTOPPULLUP_XMAX //#define ENDSTOPPULLUP_YMAX //#define ENDSTOPPULLUP_ZMAX //#define ENDSTOPPULLUP_IMAX //#define ENDSTOPPULLUP_JMAX //#define ENDSTOPPULLUP_KMAX //#define ENDSTOPPULLUP_UMAX //#define ENDSTOPPULLUP_VMAX //#define ENDSTOPPULLUP_WMAX #define ENDSTOPPULLUP_ZMIN_PROBE #endif
> 
> ..................................
> 
> // Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup). #define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop. #define I_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define J_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define K_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define U_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define V_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define W_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define I_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define J_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define K_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define U_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define V_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define W_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop. #define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe.
> 
> en Z Probe Options utilizo esto
> 
> //#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
> 
> // Force the use of the probe for Z-axis homing #define USE_PROBE_FOR_Z_HOMING Z_MIN_PROBE_PIN
> 
> ..........................................................................................
> 
> Define this pin if the probe is not connected to Z_MIN_PIN. If not defined the default pin for the selected MOTHERBOARD will be used. Most of the time the default is what you want. The simplest option is to use a free endstop connector. Use 5V for powered (usually inductive) sensors. RAMPS 1.3/1.4 boards may use the 5V, GND, and Aux4->D32 pin: For simple switches connect...
> 
> * normally-closed switches to GND and D32.
> * normally-open switches to 5V and D32.
>   */
>   #define Z_MIN_PROBE_PIN P0_10 // Pin 32 is the RAMPS defaul
>   ..............................................................................................................
>   #define FIX_MOUNTED_PROBE
> 
> ............................................................................ /**
> 
> Probe Enable / Disable The probe only provides a triggered signal when enabled. */ #define PROBE_ENABLE_DISABLE #if ENABLED(PROBE_ENABLE_DISABLE) #define PROBE_ENABLE_PIN P2_00 // Override the default pin here #endif conecte la impresora 3d a pronter face y ejecute el comando M119, si la sonda esta retraída el estado del z_probe será disparado. si aplica el comando M401 y la sonda desplegara la punta, vuelva a ejecutar el M119 y vera el estado de z_probe abierto. si esto le funciona ejecute el g28 debería funcionar la sonda. saludos

I received a document today from Biqu B1 that I would have to invert white and black wire internally, did you do that? They also sent me a firmware a little while ago, but it doesn't work!

knotart01, 2023-12-14:
> I received a document today from Biqu B1 that I would have to invert white and black wire internally, did you do that? They also sent me a firmware a little while ago, but it doesn't work!

hola como estas, he seguido las indicaciones del manual de biqu para conectar la micro probe V2, el unico problema que he tenido es que para que funcione correctamente debes desplegar la sonda con el M401 antes de hacer G28, es decir, la sonda debe estar en estado OPEN antes de iniciar la rutina de G28. he realizado HOMING, calibrar el Z OFFSET y crear una malla UBL. siempre y cuando la sonda este en estado OPEN antes de hacer HOMING. es algo raro, seguiré buscado una solución. saludos

bda101, 2023-12-24:
> I need Firmware for SKR 1.4 Biqu helpppppppppppppppppppppppppppppp

 did you find it ?

knotart01, 2023-12-28:
> > I need Firmware for SKR 1.4 Biqu helpppppppppppppppppppppppppppppp
> 
> did you find it ?
hola, esta configuración de Marlín 2.1.2  me esta funcionando para la microprobe V2 en una skr 1.4 turbo.
el conector que contiene los cables negro y blanco, debe ser conectado al zmin de la placa skr 1.4 turbo. respetando la configuración, es decir, cable negro es GND y cable blanco es SEÑAL.
[image]
[image]
[image]
[image]
[image]

CS7AFM, 2024-01-06:
> > > I need Firmware for SKR 1.4 Biqu helpppppppppppppppppppppppppppppp
> > 
> > 
> > did you find it ?
> > hola, esta configuración de Marlín 2.1.2  me esta funcionando para la microprobe V2 en una skr 1.4 turbo.
> > el conector que contiene los cables negro y blanco, debe ser conectado al zmin de la placa skr 1.4 turbo. respetando la configuración, es decir, cable negro es GND y cable blanco es SEÑAL.
> > ![endstop inverting](https://private-user-images.githubusercontent.com/153781384/293277830-80b7f504-972d-4d00-8c6d-11f969ab8ea6.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1MDkxMzMsIm5iZiI6MTcwNDUwODgzMywicGF0aCI6Ii8xNTM3ODEzODQvMjkzMjc3ODMwLTgwYjdmNTA0LTk3MmQtNGQwMC04YzZkLTExZjk2OWFiOGVhNi5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwNlQwMjQwMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01YzNjYjVhNTg5NzJiNjVjMWExODQ5MDJhNWI2MjMyZWFiMDMyY2I5ZDY1ZTFiOThmNGYyMGEwZTcxMjk3YzcwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ckqSzxWzSy-acXD8IOrwggxNpCGNXwzMk9dExbok8Io)
> > ![endstop zminprobe](https://private-user-images.githubusercontent.com/153781384/293277831-b6c7008c-c17a-45e9-85f4-9880ef2e5470.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1MDkxMzMsIm5iZiI6MTcwNDUwODgzMywicGF0aCI6Ii8xNTM3ODEzODQvMjkzMjc3ODMxLWI2YzcwMDhjLWMxN2EtNDVlOS04NWY0LTk4ODBlZjJlNTQ3MC5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwNlQwMjQwMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNzg0ODA2Mzg3MjdhNzNhMTRiZjI5NDNjYzFmZDc4ZTk4NzlkZDk3ZDgyYjFkM2Q5YTQ1YTgwYjRjMzU0ZGQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.VvN5RRKa2sD5Y_p7gXqQM5ar03GrDOZ1iR25xgLRjvs)
> > ![endstopo pullup](https://private-user-images.githubusercontent.com/153781384/293277833-f3c1923b-2e0b-406b-aae0-f38be58dcd3e.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1MDkxMzMsIm5iZiI6MTcwNDUwODgzMywicGF0aCI6Ii8xNTM3ODEzODQvMjkzMjc3ODMzLWYzYzE5MjNiLTJlMGItNDA2Yi1hYWUwLWYzOGJlNThkY2QzZS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwNlQwMjQwMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hNmY1MTc1YzNhN2U5NjY4ZTgxNTE3Yjk1NzE2NGRlNTc1ZjA4YWVlM2RiMWI3ZmVlMmNkOTA2NDc5NzA1YWNiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-pCRbVEPYAQxbGBl7nuNJgYeCD1bHC1Vr9eN6RL2AFI)
> > ![fixprobe](https://private-user-images.githubusercontent.com/153781384/293277835-f77b65d0-b8bd-4e01-8421-7a6ce2dcbc6a.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1MDkxMzMsIm5iZiI6MTcwNDUwODgzMywicGF0aCI6Ii8xNTM3ODEzODQvMjkzMjc3ODM1LWY3N2I2NWQwLWI4YmQtNGUwMS04NDIxLTdhNmNlMmRjYmM2YS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwNlQwMjQwMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xOGM0ZjU3NjVhMjFmMjZhNDY5MDc0NjE0Yjk3M2MzMjdkZDMyMjkyZGJmMjNlNTgxM2MxYTcyNmNhMzY4N2I3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.oN5mfkog06GccVOqzCTghwAxJ1ybPbzT_KhCn4Hc-Wk)
> > ![probe enable](https://private-user-images.githubusercontent.com/153781384/293277836-b614d571-63a5-4f64-b3e5-588311a94e45.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ1MDkxMzMsIm5iZiI6MTcwNDUwODgzMywicGF0aCI6Ii8xNTM3ODEzODQvMjkzMjc3ODM2LWI2MTRkNTcxLTYzYTUtNGY2NC1iM2U1LTU4ODMxMWE5NGU0NS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwNlQwMjQwMzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lOGVmMWZmODg1ODZmN2ZkYmVmNzhhNmFjNTQwNjM5NmFiNGQ2Y2VhNzdmMzYzZTRiN2Q2NWFkMjYxZGQ3NTBmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.G2KnRsY3hQbwhNbazT7tFC56qHLB-oGCOh_E2XDzNyM)

I have conected as picture
[image]

and defined this:

#define PROBE_ENABLE_DISABLE
#if ENABLED(PROBE_ENABLE_DISABLE)
  #define PROBE_ENABLE_PIN P0_10   // Override the default pin here
#endif

and this

#define Z_MIN_PROBE_PIN P2_00

as been told from Bigtreetech, but noting... :(

ElayHadad, 2024-01-25:
This [commit](https://github.com/MarlinFirmware/Marlin/commit/204de723f1c08f83e281f0629b60822b334097ed) might help you with the code modification in Marlin Firmware, or you can just compile the bugfix branch.

thisiskeithb, 2024-01-31:
Configuring the BIQU MicroProbe under Marlin just got a whole lot easier!

We now have default pins defined for most BigTreeTech, Creality, and MKS boards as well as sanity checks to ensure you've enabled the correct hit/trigger state for your probe.

Grab the latest [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) and there will now be a dedicated section for the MicroProbe:

```c
/**
 * BIQU MicroProbe
 *
 * A lightweight, solenoid-driven probe.
 * For information about this sensor https://github.com/bigtreetech/MicroProbe
 *
 * Also requires: PROBE_ENABLE_DISABLE
 */
//#define BIQU_MICROPROBE_V1  // Triggers HIGH
//#define BIQU_MICROPROBE_V2  // Triggers LOW
```

Once the next version of Marlin is released, the MicroProbe manual will be updated to reflect these new options.

Gintaras-sur, 2024-02-08:
maybe someone has the firmware ready for the microprobe?

RobertoMG9824, 2024-02-21:
I was able to configure the V2.0 microprobe with the V1.4 skr board. **Use the following procedure:**

1.- Download the latest available version of the firmware on the BIQU site. Which is **marlin 2.0.9.3**
https://github.com/bigtreetech/marlin/tree/B1-2.0.9.3.x.
2.-Follow the instructions in the README file to configure the firmware for the SKR V1.4 board.
[image]
[image]
[image]

3.-I followed the instructions in the MicroProbe V2 manual, adding an extra step:

1. Comente la linea _#Define BLTOUCH_. 
[image]

2. Comment: _#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN_
![image (https://github.com/bigtreetech/MicroProbe/assets/160571530/cb048432-8dba-4cb8-b804-7bba6679e4df)

3. Pin P0_10 is defined in _#define Z_MIN_PROBE_PIN_
![image]
(https://github.com/bigtreetech/MicroProbe/assets/160571530/2fcbbe7d-d035-496e-8f0c-470ac74b860b)

4. Pin P2_00 is defined
[image]

5. _#define FIX_MOUNTED_PROBE_
[image]

6. _#define NOZZLE_TO_PROBE_OFFSET { 0, 0, 0 }_. Yo utilice -7,-35, -1.3. I recommend that you define the value of z (In my case 1.3) at 0 and then configure it from the touch screen to find the value that works for you.
[image]

7. _#define Z_MIN_PROBE_ENDSTOP_INVERTING true_. Hay que cambiar false por true.
[image]

8. #define ENDSTOPPULLUP_ZMIN_PROBE 
[image]

4.-Upload the firmware and also upload the screen firmware (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware).

5.-Connect the probe to the back of the printer, the same as you would connect the BLTouch.

Those were the steps I followed and so far it has worked well for me. I hope they help you.

Note: To edit the firmware download visual studio code and platformIO. You can follow this reference video: https://www.youtube.com/watch?v=nCBeCZg9k0U&t=1931s

John-Vinic, 2024-02-27:
> I was able to configure the V2.0 microprobe with the V1.4 skr board. **Use the following procedure:**
> 
> 1.- Download the latest available version of the firmware on the BIQU site. Which is **marlin 2.0.9.3** https://github.com/bigtreetech/marlin/tree/B1-2.0.9.3.x. 2.-Follow the instructions in the README file to configure the firmware for the SKR V1.4 board. ![Captura de pantalla 2024-02-21 145019](https://private-user-images.githubusercontent.com/160571530/306774103-aca6552b-5cdb-40a7-b6a5-a164a45d3846.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MTAzLWFjYTY1NTJiLTVjZGItNDBhNy1iNmE1LWExNjRhNDVkMzg0Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kNmI4NzhiMTQ1NzAzZGVjODM3NGFjMjQ3MWU1MTYwNjI5ODMwMDE4M2QzZjU0OWU0N2ZiMjk1MTk1OTdiZTk4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Pj_0R0fIngLBfDekKTxErtWje21xc4eoKRP6yL3CyQ8) ![Captura de pantalla 2024-02-21 145038](https://private-user-images.githubusercontent.com/160571530/306774178-f635a32e-405d-4997-9004-fb1b8371d3af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MTc4LWY2MzVhMzJlLTQwNWQtNDk5Ny05MDA0LWZiMWI4MzcxZDNhZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MDlhMWRkNWE5YTdhMzAwYzRiOGEwZWYxY2Q3MGQ2NTBmMjlmZDBiYWVlOTZmNGU0NmYwMmVjZGQ5NGMxNzQ2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.p2-3eCsCd0DWvWE642NdjlT4Mq0mPs1OmOjDCMRuZ6s) ![Captura de pantalla 2024-02-21 145048](https://private-user-images.githubusercontent.com/160571530/306774227-82249640-6d24-46ea-a3bb-a8988c615199.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MjI3LTgyMjQ5NjQwLTZkMjQtNDZlYS1hM2JiLWE4OTg4YzYxNTE5OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03YzRjOTgzYjE3ZWNmYzY1N2M0MDE5NTZmNjcwOWEwYTIxODM4MTU3N2I5MTQwOWY3NGU5M2E1OWY0M2NlM2UxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.jgzhn9Q1rRefIVgp-tPW99g8qtajRlNBojdXcY4qFe4)
> 
> 3.-I followed the instructions in the MicroProbe V2 manual, adding an extra step:
> 
> 1. Comment: _#Define BLTOUCH_.
>    ![image](https://private-user-images.githubusercontent.com/160571530/306770387-157e8d3a-119c-4166-8c88-ed8a09874675.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcwMzg3LTE1N2U4ZDNhLTExOWMtNDE2Ni04Yzg4LWVkOGEwOTg3NDY3NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yZWJkZjAzNzY3Nzk2MmVlOTY0YTM0NzBiM2VmNDhiNzNkODQyNGI0ZmM5MmNlYmEwNDdiZGZjOGEzN2M2MjY3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.WdS1_cWOG_u1RmbJI3CRk6ahfPLFk2YcPYTeqxawMWA)
> 2. Comment: _#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN_
>    ![image](https://private-user-images.githubusercontent.com/160571530/306771504-cb048432-8dba-4cb8-b804-7bba6679e4df.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcxNTA0LWNiMDQ4NDMyLThkYmEtNGNiOC1iODA0LTdiYmE2Njc5ZTRkZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ODk1OWQ2ZjY2OTVlYTlhZDhkMzczMTIzMGUyZDc0Mjk2ODZlYmE5NzBmNjdhNWYxNjE5YmFjZWFhNDZlYTdjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.NYHl65CfbFhNyUfVM4QT1sfVABrISuHimnetCb3DDbs)
> 3. Pin P0_10 is defined in _#define Z_MIN_PROBE_PIN_
>    ![image](https://private-user-images.githubusercontent.com/160571530/306771836-2fcbbe7d-d035-496e-8f0c-470ac74b860b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcxODM2LTJmY2JiZTdkLWQwMzUtNDk2ZS04ZjBjLTQ3MGFjNzRiODYwYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wN2RjZD [... cut ...]
> 5.-Connect the probe to the back of the printer, the same as you would connect the BLTouch.
> 
> Those were the steps I followed and so far it has worked well for me. I hope that helps you.
> 
> Note: To edit the firmware download visual studio code and platformIO. You can follow this reference video: https://www.youtube.com/watch?v=nCBeCZg9k0U&t=1931s

Você precisou fazer alguma mudança nas conexões do hardware internamente?

Did you need to make any changes to the hardware connections internally?

fabianovm, 2024-02-27:
> I was able to configure the V2.0 microprobe with the V1.4 skr board. **Use the following procedure:**
>
> 1.- Download the latest available version of the firmware on the BIQU site. Which is **marlin [...]

Hello Roberto, thanks a LOT for all your hard work producing this tutorial.. I am brazilian.... Fabiano, and will resume all testing and firmware compilation I stopped trying months ago...

Will intall microprobe again and do step by step all you said... will take a week, have few time, but I will give it a try, for sure!!!

One important detail... in fact a question... when I was trying to make it work months ago, I flashed the  screen firmware (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware) and tried to activate the microprobe. As soon I navigate to ABL in TFT touch, it gives me error, because I understood, that ABL section in screen firmware was to the BLTOUCH.

Now the question... with this tutorial, flashing both firmware, skr1.4 and touchscreen, will I have access and control the ABL functions in the ABL section of touchscreen? I use the modern interface, not marlin interface, so this is important for me. Let me know if the ABL section of screen firmware is now prepared and functional for the microprobe instead of bltouch. 

Again, thanks a LOT!

RobertoMG9824, 2024-02-28:
> > Pude configurar la microsonda V2.0 con la placa skr V1.4. **Utilice el siguiente procedimiento:**
> > 1.- Descargue la última versión disponible del firmware en el sitio de BIQU. Que es **marlin 2.0.9.3 ** https://github.com/bigtreetech/marlin/tree/B1-2.0.9.3.x . 2.-Siga las instrucciones del archivo README para configurar el firmware de la placa SKR V1.4.![Captura de pantalla 2024-02-21 145019](https://private-user-images.githubusercontent.com/160571530/306774103-aca6552b-5cdb-40a7-b6a5-a164a45d3846.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MTAzLWFjYTY1NTJiLTVjZGItNDBhNy1iNmE1LWExNjRhNDVkMzg0Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kNmI4NzhiMTQ1NzAzZGVjODM3NGFjMjQ3MWU1MTYwNjI5ODMwMDE4M2QzZjU0OWU0N2ZiMjk1MTk1OTdiZTk4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Pj_0R0fIngLBfDekKTxErtWje21xc4eoKRP6yL3CyQ8) ![Captura de pantalla 2024-02-21 145038](https://private-user-images.githubusercontent.com/160571530/306774178-f635a32e-405d-4997-9004-fb1b8371d3af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MTc4LWY2MzVhMzJlLTQwNWQtNDk5Ny05MDA0LWZiMWI4MzcxZDNhZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MDlhMWRkNWE5YTdhMzAwYzRiOGEwZWYxY2Q3MGQ2NTBmMjlmZDBiYWVlOTZmNGU0NmYwMmVjZGQ5NGMxNzQ2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.p2-3eCsCd0DWvWE642NdjlT4Mq0mPs1OmOjDCMRuZ6s) ![Captura de pantalla 2024-02-21 145048](https://private-user-images.githubusercontent.com/160571530/306774227-82249640-6d24-46ea-a3bb-a8988c615199.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MjI3LTgyMjQ5NjQwLTZkMjQtNDZlYS1hM2JiLWE4OTg4YzYxNTE5OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03YzRjOTgzYjE3ZWNmYzY1N2M0MDE5NTZmNjcwOWEwYTIxODM4MTU3N2I5MTQwOWY3NGU5M2E1OWY0M2NlM2UxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.jgzhn9Q1rRefIVgp-tPW99g8qtajRlNBojdXcY4qFe4)
> > 3.-Seguí las instrucciones del manual de MicroProbe V2, añadiendo un paso extra:
> > 
> > 1. Comentario: _#Defina BLTOUCH_ .
> >    ![imagen](https://private-user-images.githubusercontent.com/160571530/306770387-157e8d3a-119c-4166-8c88-ed8a09874675.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcwMzg3LTE1N2U4ZDNhLTExOWMtNDE2Ni04Yzg4LWVkOGEwOTg3NDY3NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yZWJkZjAzNzY3Nzk2MmVlOTY0YTM0NzBiM2VmNDhiNzNkODQyNGI0ZmM5MmNlYmEwNDdiZGZjOGEzN2M2MjY3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.WdS1_cWOG_u1RmbJI3CRk6ahfPLFk2YcPYTeqxawMWA)
> > 2. Comentario: _#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN_
> >    ![imagen](https://private-user-images.githubusercontent.com/160571530/306771504-cb048432-8dba-4cb8-b804-7bba6679e4df.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcxNTA0LWNiMDQ4NDMyLThkYmEtNGNiOC1iODA0LTdiYmE2Njc5ZTRkZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ODk1OWQ2ZjY2OTVlYTlhZDhkMzczMTIzMGUyZDc0Mjk2ODZlYmE5NzBmNjdhNWYxNjE5YmFjZWFhNDZlYTdjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.NYHl65CfbFhNyUfVM4QT1sfVABrISuHimnetCb3DDbs)
> > 3. El pin P0_10 está definido en _#define Z_MIN_PROBE_PIN_
> >    ![imagen](https://private-user-images.githubusercontent.com/160571530/306771836-2fcbbe7d-d035-496e-8f0c-470ac74b860b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNTUwOTIsIm5iZiI6MTcwOTA1NDc5MiwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcxODM2LTJmY2JiZTdkLWQwMzUtNDk2ZS04ZjBjLTQ3MGFjNzRiODYwYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QxNzI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW [... cut ...]
> > Nota: Para editar el firmware, descargue el código de Visual Studio y platformIO. Puedes seguir este vídeo de referencia: https://www.youtube.com/watch?v=nCBeCZg9k0U&t=1931s
> 
> ¿Tienes que cambiar algo de conexión con el hardware internamente?
> 
> ¿Necesitó realizar algún cambio en las conexiones de hardware internamente?

No, It isn't necessary change anything in the internal conexion. I connected directly the sensor in the BLTouch port. The microprobe V2.0 is ready for SKR V1.4. 
[image]

RobertoMG9824, 2024-02-28:
> > I was able to configure the V2.0 microprobe with the V1.4 skr board. **Use the following procedure:**
> > 1.- Download the latest available version of the firmware on the BIQU site. Which is **marlin 2.0.9.3** https://github.com/bigtreetech/marlin/tree/B1-2.0.9.3.x. 2.-Follow the instructions in the README file to configure the firmware for the SKR V1.4 board. ![Captura de pantalla 2024-02-21 145019](https://private-user-images.githubusercontent.com/160571530/306774103-aca6552b-5cdb-40a7-b6a5-a164a45d3846.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNzc4OTMsIm5iZiI6MTcwOTA3NzU5MywicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MTAzLWFjYTY1NTJiLTVjZGItNDBhNy1iNmE1LWExNjRhNDVkMzg0Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QyMzQ2MzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00YzNjM2U2NzEwNDRiYzlkZTZlMGE5Y2UyOTE3NGRmMjM5MDFhMWExNTMzYWMyNzUyNmYzMDIwNjZiNGQ5MjBhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.7_j0YUZ9GQuP8umPujvyvge_OIrbBAasYt_8qwk5kYM) ![Captura de pantalla 2024-02-21 145038](https://private-user-images.githubusercontent.com/160571530/306774178-f635a32e-405d-4997-9004-fb1b8371d3af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNzc4OTMsIm5iZiI6MTcwOTA3NzU5MywicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MTc4LWY2MzVhMzJlLTQwNWQtNDk5Ny05MDA0LWZiMWI4MzcxZDNhZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QyMzQ2MzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NjEwMjVlZTE3ODc1NWU2NzMyNWQ1ZTM3ODM0NDY2MmMwNmViNTQzMDNiZjRiZWJiZmY2ZGI0ZTNhOTQxYjkzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ENWTiOqnWoVqpV5mT--nowdp2ef7lKs91tTrwfyiy40) ![Captura de pantalla 2024-02-21 145048](https://private-user-images.githubusercontent.com/160571530/306774227-82249640-6d24-46ea-a3bb-a8988c615199.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNzc4OTMsIm5iZiI6MTcwOTA3NzU5MywicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MjI3LTgyMjQ5NjQwLTZkMjQtNDZlYS1hM2JiLWE4OTg4YzYxNTE5OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QyMzQ2MzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MjMyZTk2NGNmNjQ1OGM5MmM2N2UyODA4NjljY2I5MWU3NWY5MDFiM2FlOGQwNmU5NTFlNzY1YzNmYmE2ODczJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.RcqH196RsjCrWO7iYYLDNjT2TRM1ypYpG4zY-WYDzh0)
> > 3.-I followed the instructions in the MicroProbe V2 manual, adding an extra step:
> > 
> > 1. Comment: _#Define BLTOUCH_.
> >    ![image](https://private-user-images.githubusercontent.com/160571530/306770387-157e8d3a-119c-4166-8c88-ed8a09874675.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNzc4OTMsIm5iZiI6MTcwOTA3NzU5MywicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcwMzg3LTE1N2U4ZDNhLTExOWMtNDE2Ni04Yzg4LWVkOGEwOTg3NDY3NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QyMzQ2MzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YzA2MzU1MmE0ZTdlYTMyZGIwOGY2ODkyNjg5ZmFhODI1Njk3ODY1YzQyYjdmZTc4NzliZmFhYzQxZmFmZGEyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.r57p6ca2ncmrebAs-SKzwflYmeLn80DZlpHrLh5z3sA)
> > 2. Comment: _#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN_
> >    ![image](https://private-user-images.githubusercontent.com/160571530/306771504-cb048432-8dba-4cb8-b804-7bba6679e4df.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNzc4OTMsIm5iZiI6MTcwOTA3NzU5MywicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcxNTA0LWNiMDQ4NDMyLThkYmEtNGNiOC1iODA0LTdiYmE2Njc5ZTRkZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QyMzQ2MzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNmY0MWM2NTU5OTYwOWE1MjQyMzA4NzliNmEzOGZmZjRkNDE3YzA5MTVhM2ZiZmYzODZiNDQ4ZDVhZjRhZGQ5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.gwZixRKagsxkwfZpSzGSiP9R1o87CQqPZ_iulkC1DX4)
> > 3. Pin P0_10 is defined in _#define Z_MIN_PROBE_PIN_
> >    ![image](https://private-user-images.githubusercontent.com/160571530/306771836-2fcbbe7d-d035-496e-8f0c-470ac74b860b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNzc4OTMsIm5iZiI6MTcwOTA3NzU5MywicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcxODM2LTJmY2JiZTdkLWQwMzUtNDk2ZS04ZjBjLTQ3MGFjNzRiODYwYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyN1QyMzQ2MzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25h [... cut ...]  is now prepared and functional for the microprobe instead of bltouch.
> 
> Again, thanks a LOT!

@fabianovm Thank you very much, I really hope that I can help you with the improvised tutorial. I'm Mexican, nice to meet you bro.
Answering your question: Yes. You can control the microprobe from the bigthreetech interface. I show you the functions it provides.
[image]

I have been able to carry out all the configuration with these tools and I have had no problems.

Good luck with the installation!

fabianovm, 2024-02-28:
> I was able to configure the V2.0 microprobe with the V1.4 skr board. **Use the following procedure:**
> 
> 1.- Download the latest available version of the firmware on the BIQU site. Which is **marlin 2.0.9.3**
> https://github.com/bigtreetech/marlin/tree/B1-2.0.9.3.x.
> 2.-Follow the instructions in the README file to configure the firmware for the SKR V1.4 board.
> [image]
> [image]
> [image]
> 
> 
> 3.-I followed the instructions in the MicroProbe V2 manual, adding an extra step:
> 
> 1. Comment: _#Define BLTOUCH_. 
> [image]
> 
> 2. Comment: _#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN_
> [image]
> 
> 3. Pin P0_10 is defined in _#define Z_MIN_PROBE_PIN_
> [image]
> 
> 4. Pin P2_00 is defined
> [image]
> 
> 5. _#define FIX_MOUNTED_PROBE_
> [image]
> 
> 6. _#define NOZZLE_TO_PROBE_OFFSET { 0, 0, 0 }_. I use (-7,-35, -1.3). I recommend that you define the value of z (In my case -1.3) at 0 and then configure it from the touch screen to find the value that works for you.
> [image]
> 
> 
> 7. _#define Z_MIN_PROBE_ENDSTOP_INVERTING true_. Change false to true.
> [image]
> 
> 8. #define ENDSTOPPULLUP_ZMIN_PROBE 
> [image]
> 
> 
> 4.-Upload the firmware and also upload the screen firmware (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware).
> 
> 5.-Connect the probe to the back of the printer, the same as you would connect the BLTouch.
> 
> Those were the steps I followed and so far it has worked well for me. I hope that helps you.
> 
> 
> Note: To edit the firmware download visual studio code and platformIO. You can follow this reference video: https://www.youtube.com/watch?v=nCBeCZg9k0U&t=1931s

@RobertoMG9824 , thanks, one more question.... in your tutorial, on procedure "3.-I followed the instructions in the MicroProbe V2 manual, adding..." you meandros this changes in following pictures?![Screenshot_20240228_070057_GitHub.jpg](https://github[image]

3e-44d9-8b00-40ab3b6ec621)

[image]

fabianovm, 2024-02-28:
[image]

fabianovm, 2024-02-28:
[image]

fabianovm, 2024-02-28:
Thanks bro, as soon as I get some time , will work on that! Even if it take more time, I ll let you know the results... thanks thanks!

fabianovm, 2024-02-28:
> I was able to configure the V2.0 microprobe with the V1.4 skr board. **Use the following procedure:**
>
> 1.- Download the latest available version of the firmware on the BIQU site. Which is **marlin [...]

@RobertoMG9824  did you used BIGTREE_GD_TFT35_V3.0_B1.27.x.bin or BIGTREE_TFT35_V3.0_B1.27.x.bin  for the LCD firmware? Do you know the difference of this GD version? I am already using BIGTREE_TFT35_V3.0_B1.27.x.bin for months, should I stay with it or go for the BIGTREE_GD_TFT35_V3.0_B1.27.x.bin? Thanks bro!!

RobertoMG9824, 2024-02-28:
@

> > Pude configurar la microsonda V2.0 con la placa skr V1.4. **Utilice el siguiente procedimiento:**
> > 1.- Descargue la última versión disponible del firmware en el sitio de BIQU. Que es **marlin 2.0.9.3 ** https://github.com/bigtreetech/marlin/tree/B1-2.0.9.3.x . 2.-Siga las instrucciones del archivo README para configurar el firmware de la placa SKR V1.4.![Captura de pantalla 2024-02-21 145019](https://private-user-images.githubusercontent.com/160571530/306774103-aca6552b-5cdb-40a7-b6a5-a164a45d3846.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxNDQ0MDQsIm5iZiI6MTcwOTE0NDEwNCwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MTAzLWFjYTY1NTJiLTVjZGItNDBhNy1iNmE1LWExNjRhNDVkMzg0Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyOFQxODE1MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xODg1YjMzMWZmYzgyNjEwMjU2YzU3MTk0MDJjNmY2YTc3OTRjOGZjNGNjMjM0ZDQzYTk2NTFjZmY4NWRlZTFlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.7t2ejEx0THeGPoqke4nnwAmgoyT4SiJzND13XVh_dQ0) ![Captura de pantalla 2024-02-21 145038](https://private-user-images.githubusercontent.com/160571530/306774178-f635a32e-405d-4997-9004-fb1b8371d3af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxNDQ0MDQsIm5iZiI6MTcwOTE0NDEwNCwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MTc4LWY2MzVhMzJlLTQwNWQtNDk5Ny05MDA0LWZiMWI4MzcxZDNhZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyOFQxODE1MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02Mjc3NTU3MmI3ZDE0MDUyMTc3MTc4MjBhZDY4YjI4OWYxZWU5ZDZiNTBkYTcwOTMxZWQ3YjdmMTBmYzlhMDkzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.PUkl5tktb0rAmXU6PfY04G6kjehrksDkcutrZ9EJByA) ![Captura de pantalla 2024-02-21 145048](https://private-user-images.githubusercontent.com/160571530/306774227-82249640-6d24-46ea-a3bb-a8988c615199.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxNDQ0MDQsIm5iZiI6MTcwOTE0NDEwNCwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2Nzc0MjI3LTgyMjQ5NjQwLTZkMjQtNDZlYS1hM2JiLWE4OTg4YzYxNTE5OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyOFQxODE1MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hMTBjZmUxNTk3NTI2MDQ3MWU2ODE3ZGRiYjZiMjI3MTA3MzViNTE3MzA4MmViYTgyMWFmODY0NWMwNGFiODZhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.MNOF_JdkoPs6oe-7u52D9TugxZAMn69xcpAMxiRDi2A)
> > 3.-Seguí las instrucciones del manual de MicroProbe V2, añadiendo un paso extra:
> > 
> > 1. Comentario: _#Defina BLTOUCH_ .
> >    ![imagen](https://private-user-images.githubusercontent.com/160571530/306770387-157e8d3a-119c-4166-8c88-ed8a09874675.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxNDQ0MDQsIm5iZiI6MTcwOTE0NDEwNCwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcwMzg3LTE1N2U4ZDNhLTExOWMtNDE2Ni04Yzg4LWVkOGEwOTg3NDY3NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyOFQxODE1MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NDRhMDc3ZWRkNGUyYTllYWIyMzQ0NmQ5M2EwZGI2YWQ3MDVhNjcxOWNjYTFhOTNiOTY0MTZlMDYxN2QzMGM2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.HtRgoU9pj0vbZQ2BvqpEJJj4WTVyqkKDEjRWk4ZPlY0)
> > 2. Comentario: _#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN_
> >    ![imagen](https://private-user-images.githubusercontent.com/160571530/306771504-cb048432-8dba-4cb8-b804-7bba6679e4df.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxNDQ0MDQsIm5iZiI6MTcwOTE0NDEwNCwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcxNTA0LWNiMDQ4NDMyLThkYmEtNGNiOC1iODA0LTdiYmE2Njc5ZTRkZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyOFQxODE1MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYzJhMDEzZGZkYTQ0ZTZlNjRiNGUzZjk4OTYxMjJjMWU3MjM2Y2ViNWQyY2ZlN2Q3ZmFjNGQxZTNiZDMyOGUwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.yTKQFLRWKqw8cPXx3MaV6LA3D_Z9jvnnhXrFMYuwKJ8)
> > 3. El pin P0_10 está definido en _#define Z_MIN_PROBE_PIN_
> >    ![imagen](https://private-user-images.githubusercontent.com/160571530/306771836-2fcbbe7d-d035-496e-8f0c-470ac74b860b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxNDQ0MDQsIm5iZiI6MTcwOTE0NDEwNCwicGF0aCI6Ii8xNjA1NzE1MzAvMzA2NzcxODM2LTJmY2JiZTdkLWQwMzUtNDk2ZS04ZjBjLTQ3MGFjNzRiODYwYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyOFQxODE1MDRaJlgtQW16LUV4cGlyZXM9MzAwJlg [... cut ...]
> @RobertoMG9824 ¿Utilizó BIGTREE_GD_TFT35_V3.0_B1.27.x.bin o BIGTREE_TFT35_V3.0_B1.27.x.bin para el firmware LCD? ¿Conoces la diferencia de esta versión GD? Ya estoy usando BIGTREE_TFT35_V3.0_B1.27.x.bin durante meses, ¿debería seguir con él o optar por BIGTREE_GD_TFT35_V3.0_B1.27.x.bin? ¡¡Gracias hermano!!

@fabianovm I'm using BIGTREE_TFT35_V3.0_B1.27.x.bin. I'm not sure what the difference is between the versions. I read that it depends on the type of boards, the newer ones use the GD firmware, but I am not sure about this information. 

Bro, I recommend you that you should stay using BIGTREE_TFT35_V3.0_B1.27.x.bin.

fabianovm, 2024-02-28:
@RobertoMG9824 , thanks, I will not touch the lcd firmware, it is ok. But do not laugh on me...I forgot where to click to obtain the marlin firmware, th3 ,zip, on this page.... can you help?

[image]

fabianovm, 2024-02-28:
@RobertoMG9824 , I found little time and made all changes in marlin 2.0.9.3.x as following your tutorial....
Run build for skr1.4 and no matter what I receive a lot of warnings:
PS D:\USER\Fabiano\3D Print\BIQU B1\FIRMWARE\MARLIN\Marlin-B1-2.0.9.3.x_PROBE> platformio run --silent -e LPC1768
Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please 
use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Marlin\src\inc\Warnings.cpp:550:4: warning: #warning "Driver DIAG pins must be physically removed unless SENSORLESS_HOMING is 
enabled. (See https://bit.ly/2ZPRlt0)" [-Wcpp]
  550 |   #warning "Driver DIAG pins must be physically removed unless SENSORLESS_HOMING is enabled. (See https://bit.ly/2ZPRlt0)"
      |    ^~~~~~~
PS D:\USER\Fabiano\3D Print\BIQU B1\FIRMWARE\MARLIN\Marlin-B1-2.0.9.3.x_PROBE> echo "done" >"C:\Users\FVM\AppData\Local\Temp\ipc"
PS D:\USER\Fabiano\3D Print\BIQU B1\FIRMWARE\MARLIN\Marlin-B1-2.0.9.3.x_PROBE>

The firmware.bin is generated. Is that OK, it happened all this warnings for you? What do you think, can I flash even with this warnings? Is there a way to remove them and have a totally clean build?

thanks Bro! Fabiano

RobertoMG9824, 2024-02-28:
> @RobertoMG9824 , I found little time and made all changes in marlin 2.0.9.3.x as following your tutorial.... Run build for skr1.4 and no matter what I receive a lot of warnings: PS D:\USER\Fabiano\3D Print\BIQU B1\FIRMWARE\MARLIN\Marlin-B1-2.0.9.3.x_PROBE> platformio run --silent -e LPC1768 Warning! Ignore unknown configuration option `monitor_flags` in section [env] Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead Warning! `src_build_flags` configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please use `build_src_flags` instead Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `build_src_filter` instead Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead Marlin\src\inc\Warnings.cpp:550:4: warning: #warning "Driver DIAG pins must be physically removed unless SENSORLESS_HOMING is enabled. (See https://bit.ly/2ZPRlt0)" [-Wcpp] 550 | #warning "Driver DIAG pins must be physically removed unless SENSORLESS_HOMING is enabled. (See https://bit.ly/2ZPRlt0)" | ^~~~~~~ PS D:\USER\Fabiano\3D Print\BIQU B1\FIRMWARE\MARLIN\Marlin-B1-2.0.9.3.x_PROBE> echo "done" >"C:\Users\FVM\AppData\Local\Temp\ipc" PS D:\USER\Fabiano\3D Print\BIQU B1\FIRMWARE\MARLIN\Marlin-B1-2.0.9.3.x_PROBE>
> 
> The firmware.bin is generated. Is that OK, it happened all this warnings for you? What do you think, can I flash even with this warnings? Is there a way to remove them and have a totally free of messages or errors firmware.bin build?
> 
> thanks Bro! Fabiano

@fabianovm I got the same warnings and I had no problems with the printer. In this case, I consider that the warnings can be ignored. The warnings only make references that in future updates the src_filter configuration will no longer be used. You could change the src_filter configuration to build_src_filter, but you could get into problems with the board's bios.

If you have no errors in the code and only have warnings, you can continue with the flash.

I'm here for any other questions you may have, good luck bro!

fabianovm, 2024-02-28:
@RobertoMG9824 , thanks, I will not touch the lcd firmware, it is ok. But do not laugh on me...I forgot where to click to obtain the marlin firmware, th3 ,zip, on this page.... can you help?

[image]

@RobertoMG9824 , thanks again, perfect... just have to wait 11 hours to do it because the printer is printing a big piece since yesterday.., I ll flash it and call you for help and or feedback for sure! Really,  thank you.

fabianovm, 2024-03-02:
@RobertoMG9824 how are you bro? I just flashed the printer... and......... IT IS WORKING! I still adjusting babystep, things like that, zi have never had a autobed leveling.... if is there some basics to do kindly tell me, but I already clicked on print the test cube.stl and the printer did the autol3v3l and started printing... far away from the bed, but I am working on that... I let you know when it is perfect!

Thanks 

Thanks

Thanks

And 

Many thanks!

fabianovm, 2024-03-02:
@RobertoMG9824 how are you bro? I just flashed the printer... and......... IT IS WORKING! I still adjusting babystep, things like that, zi have never had a autobed leveling.... if is there some basics to do kindly tell me, but I already clicked on print the test cube.stl and the printer did the autol3v3l and started printing... far away from the bed, but I am working on that... I let you know when it is perfect!

One thing I noticed is that when the probe is measuring in the bed, it starts more to the right than expected and the last measures almost at the left end of bed...how adjust that?

Thanks 

Thanks

Thanks

And 

Many thanks!

fabianovm, 2024-03-03:
@thisiskeithb Hello Keith, I saw the two microprobe references on bugfix version, but just the declaration, nothing else implemented. Do you have any info regarding when, finally, after years, biqu will release firmware that works microprobe v2 TOTALLY, really, without eventual fails, with complete TFT control, stow, deploy, and everything oriented to the microprobe?? Regards!

thisiskeithb, 2024-03-03:
> @thisiskeithb Hello Keith, I saw the two microprobe references on bugfix version, but just the declaration, nothing else implemented. Do you have any info regarding when, finally, after years, biqu [...]

It’s a bit more than just a couple of declarations 😀. Probe deploy/stow, etc. work.

You can view all the changes in https://github.com/MarlinFirmware/Marlin/commit/204de723f1c08f83e281f0629b60822b334097ed. This will be in `2.1.3`

CS7AFM, 2024-03-03:
I should have stopped and send to trash this new-non-working-probe when I had the time... After following this procedure my printer went nuts, funny noises from motors, skips some parts on printing... 
To TS I made eprom reset, tried to go back to it's original firmware without success.

fabianovm, 2024-03-03:
> I should have stopped and send to trash this new-non-working-probe when I had the time... After following this procedure my printer went nuts, funny noises from motors, skips some parts on printing... 
> To TS I made eprom reset, tried to go back to it's original firmware without success.

Hi, I understand you, totally. This try and error costed me scratches on the perfect  bed, plus a new nozzle. Thanks biqu. 

I almost smashed probe some months ago, but decided to wait. I think this will be solved just when biqu releases marlin 2.1.3 I think, in some years...lol...an absurd to sell a hardware without driver or firmware or even detailed instructions...

Suggestion:
If you printer is turning on and you have access to menu, find the option eprom and reset it. And flash the firmware you used to have and work, without microprobe. After flash is complete, turn off printer, remove scarf and turn on printer.... go again in eprom option and click reset again... turn off printer. It should work again I think, I am not a specialist on that, just want to help. Regards!

fabianovm, 2024-03-03:
> It’s a bit more than just a couple of declarations 😀. Probe deploy/stow, etc. work.
>
> You can view all the changes in [...]

it is an absurd to launch microprobe and firmware years ahead... I love my B1, but what biqu did with microprobe, letting everybody crazy trying to make it work, without proper tutorial, damaging printers and firmwares, is unforgivable in my honest opinion , do not take as personal you just want to help, know you from other topics, nice guy.

Hope someday, in fact, in next biqu marlin firmware this is really and totally solved. And when it arrives the printer still works. Best regards!

CS7AFM, 2024-03-03:
Hi guys! @fabianovm I went for another try (reset, new build, reset, new build, reset, new build... ) and it finally worked... thanks all for your sharing and thanks for @RobertoMG9824 tutorial.

This process was a great lesson learn, that the post-sell from this company is not what I would expect.

Best regards!

bda101, 2024-03-05:
If i can ask, can any1 share their config and adv config files ?  i would like to compare with what i have, at the moment i have an issue when mine stays deployed after homing... and hits the bed after the purge line...

John-Vinic, 2024-03-10:
@RobertoMG9824  It worked, but the tip of the touch is touching the table and bending when moving, it almost broke, what should I do?

jonathanmwagstaff, 2024-03-15:
> 4.-Upload the firmware and also upload the screen firmware (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware).

I did the steps before and I think that I have done them right. I have never uploaded screen firmware before. Am I supposed to upload a precompiled firmware for the screen?

Pascalmarin, 2024-04-18:
Hello @RobertoMG9824,
thank you very much for your work. I did the entire installation according to your guide, I compiled and installed the firmware on my machine, I also updated the firmware of my TFT, but when booting I don't see any menu about autoleveling, I can't find it anywhere. Is it possible that something else needs to be modified in the FW for this menu to appear? When I turn on the microprobe, it extends and retracts the PIN four times.
Thanks and regards from Spain

Pascalmarin, 2024-04-18:
Mi placa es la SKR 1.4 y mi versión de MArlin 2.1.2.2. Esta montada en una artillery genius.

AiSuperBot, 2024-05-03:
> I am encountering the following error when I try to update the suggested firmware error : EEPROM CRC Mismatch - (stored) 38091 != 63706 (calculated)

Update [TFT Firmware](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware?tab=readme-ov-file#firmware-update) (Yes, even if you have latest. Download the whole repo follow [this](https://www.youtube.com/watch?v=nCBeCZg9k0U&t=1931s @RobertoMG9824 shared).

AiSuperBot, 2024-05-03:
Hi there @RobertoMG9824. Here is my configuration [Marlin-B1-2.0.9.3.x - Configuration.zip](https://github.com/bigtreetech/MicroProbe/files/15201497/Marlin-B1-2.0.9.3.x.-.Configuration.zip) (changed pins for extruder cause the default one stopped working at some powerloss we had).
For some reason I get `Unknown command: "M280 P0 S10"` and when I try to use UBL it goes crazy.
Help would be much appreciated! Thanks a lot!

AiSuperBot, 2024-05-04:
Microprobe V2 does not retract when finished probing, hence crashing into the bed. Anyone knows how to fix?
Thanks a lot for your time, help is much appreciated!

tuliosciotta, 2024-05-15:
Does anyone have an builded B1 SKR1.4 firmware to share?  One that makes the microprobe work...

bda101, 2024-05-18:
Does any1 have a full firmware to share ? for SKR 1.4 ?

SkinnyV, 2024-05-29:
Man, this is crazy, that thing has been out for over a year and is advertised as supporting the BIQU B1, yet no firmware, instruction or any kind of official information on how to use it with this printer. It's made by the same company, not like I'm buying a third party-probe and expecting it to work right away. Compiling firmware, are you kidding me? Where is the official firmware for this thing, this is super ridiculous.

Dementran99, 2024-09-28:
i have an ender 3 white an skr 1.4 when i am hoing it stops mid air and say probing failiur the probe works now but how do i do so it probs the bed even when i have it tuch the bed and homing then it lift about 15mm and the try to probe and stop 11mm in the air

GSVil, 2024-11-24:
For those interested in downloading the working firmware, I have uploaded my BIN file. It took me almost a week to get it working correctly, recompiling and testing multiple versions of MARLIN before eventually settling on Marlin 2.1.2.5. I believe this should work great for everyone, but please let me know if something is not working or needs adjusting.

**Please take note of the following points before you use my BIN file:**

1) The files provided will only work if your BIQU-B1 printer is equipped with the **BOARD_BTT_SKR_V1_4** Motherboard.
2) Make sure you have flashed your BIQU-B1 with the latest BIGTREETECH TouchScreen Firmware; otherwise, it might not be compatible with this motherboard firmware.
3) The "MICROPROBE-V2" version of the firmware assumes you have used the probe connector at the back of the BIQU-B1.
4) Using the "MICROPROBE-V2" version of the firmware will **disable** the **Z-axis micro-switch**, and the MICROPROBE-V2 will be used for Z-Homing and leveling.
5) Make sure you adjust the **Probe Offsets** for your specific installation; mine is installed to the right of the extruder.
6) **Mesh-Bed-Leveling** will be replaced with **Auto Bed Leveling**, so make sure to enable ABL in CURA or whichever slicer you use.

https://github.com/GSVil/BIQU-B1-SKR-V1.4---Firmware-BIN-Files/tree/main

Dementran99, 2024-11-24:
Thanks for the files and the help
I will test it now

Skickat från Outlook för Android<https://aka.ms/AAb9ysg>

https://github.com/GSVil/BIQU-B1-SKR-V1.4---Firmware-BIN-Files/tree/main

ETorenvliet, 2024-12-09:
First of all thanks for the work!!
Are you willing to share the config files? I created a bracket for my sensor at the right of the hotend to prevent blocking the front cooling fan.

Dementran99, 2024-12-09:
Thanks for the info I just converted my marlin printer to klipper now it works

Skickat från Outlook för Android<https://aka.ms/AAb9ysg>

https://github.com/GSVil/BIQU-B1-SKR-V1.4---Firmware-BIN-Files/tree/main

bda101, 2025-01-10:
> No, It isn't necessary change anything in the internal conexion. I connected directly the sensor in the BLTouch port. The microprobe V2.0 is ready for SKR V1.4. [...]

 Hello,
Could you perhaps share the config files that you used ?
My B1 has dual z each with its own driver.. so i would need to modify it a bit.
Thank You

bda101, 2025-01-10:
> For those interested in downloading the working firmware, I have uploaded my BIN file. It took me almost a week to get it working correctly, recompiling and testing multiple versions of MARLIN before eventually settling on Marlin 2.1.2.5. I believe this should work great for everyone, but please let me know if something is not working or needs adjusting.
> 
> **Please take note of the following points before you use my BIN file:**
> 
>     1. The files provided will only work if your BIQU-B1 printer is equipped with the **BOARD_BTT_SKR_V1_4** Motherboard.
> 
>     2. Make sure you have flashed your BIQU-B1 with the latest BIGTREETECH TouchScreen Firmware; otherwise, it might not be compatible with this motherboard firmware.
> 
>     3. The "MICROPROBE-V2" version of the firmware assumes you have used the probe connector at the back of the BIQU-B1.
> 
>     4. Using the "MICROPROBE-V2" version of the firmware will **disable** the **Z-axis micro-switch**, and the MICROPROBE-V2 will be used for Z-Homing and leveling.
> 
>     5. Make sure you adjust the **Probe Offsets** for your specific installation; mine is installed to the right of the extruder.
> 
>     6. **Mesh-Bed-Leveling** will be replaced with **Auto Bed Leveling**, so make sure to enable ABL in CURA or whichever slicer you use.
> 
> 
> https://github.com/GSVil/BIQU-B1-SKR-V1.4---Firmware-BIN-Files/tree/main
> 
> ![20241125_095951](https://private-user-images.githubusercontent.com/69642887/390310276-d661a207-dc09-4060-8903-a7f8ac099967.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY1MjA3NDIsIm5iZiI6MTczNjUyMDQ0MiwicGF0aCI6Ii82OTY0Mjg4Ny8zOTAzMTAyNzYtZDY2MWEyMDctZGMwOS00MDYwLTg5MDMtYTdmOGFjMDk5OTY3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEwVDE0NDcyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFiZWEwYjA3NTNmYzkzYjA0MDQ3ZGU3NGIzODExOTFkNTU0Yjc0YjlhNzExYTM2ZGZhMTYzM2FmNWUwYTBiNDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.QrxZke1s4E1xb5iibWhbjgDGGp3LYtwbqyBahjh3p20)
> 
> ![20241124_230826](https://private-user-images.githubusercontent.com/69642887/390310735-fad0d36b-2765-4608-a6a9-7edec462dbe4.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY1MjA3NDIsIm5iZiI6MTczNjUyMDQ0MiwicGF0aCI6Ii82OTY0Mjg4Ny8zOTAzMTA3MzUtZmFkMGQzNmItMjc2NS00NjA4LWE2YTktN2VkZWM0NjJkYmU0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEwVDE0NDcyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMwMGViYTE0ODQ3ODdmNTM0NGUxNTNiMGZkMWMyODNlNjU4YzNhNjRmZjFiOTc5YzE3N2IxN2ZjYTMzYTNhODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.o6248oT8am4HmGMELsltbfD86EPYx-xHouTcFZbqgq4)

Could you share the config files if possible ?

[2 more replies]