bigtreetech/MicroProbe #14: Stock ender 3 4.2.2 board

Birddog1148, 2023-10-01:
Am I wrong to think that this thing will work on a 4.2.2 board?  For the life of me cant get this to work and my searches are not getting good results.

Egosumumbravir, 2023-10-24:
Same boat. Ender 3v2, v4.2.2 silent board; MRiscoC Professional Firmware. Works as expected with a BL-Touch probe.

Microprobe cycles 3x at boot-up, but them won't deploy when needed. Have noted it seems to randomly deploy/retract as I much about in various configuration screens.

delong369, 2023-10-28:
> Same boat. Ender 3v2, v4.2.2 silent board; MRiscoC Professional Firmware. Works as expected with a BL-Touch probe.
> 
> Microprobe cycles 3x at boot-up, but them won't deploy when needed. Have noted it seems to randomly deploy/retract as I much about in various configuration screens.

have you figured this out? i have the 4.2.2 and am trying to get it to work to

Egosumumbravir, 2023-10-29:
> have you figured this out? i have the 4.2.2 and am trying to get it to work to

Still a work in progress, but getting there. I am now understanding that this sensor might try and use the same plug as a BL/CR-Touch but it's not a clone of them and requires changes to code and compiling a new firmware.

delong369, 2023-10-29:
I went ahead and put a bl touch clone on there and it works good but I'd
like to use the bq micro on my pro or my ender extender pro. Wonder if you
can just ad firmware to the board for it with out messing with the firmware
already on the board

Egosumumbravir, 2023-10-31:
Sadly, there's only one option - flash a suitable system firmware compiled with settings appropriate to the device.
I'm working with MRiscoC Professional rather than original Marlin.

Egosumumbravir, 2023-11-06:
UPDATE: I have the Microprobe v2 working as expected with a custom compiled MRiscoC Professional Firmware.
https://github.com/mriscoc/Ender3V2S1

delong369, 2023-11-06:
I'm putting mine on an ender 3 pro with the 4.2.7 board and 400x400 build
plate. It says mriscoc isn't compatible with the pro.  I do have a v2 with
mriscoc firmware on it and it has a bl touch on it. But since my pro has
such a big build plate and the biqu microprobe is more precise with
leveling I wanted it on there. If it comes down to it I may swap
motherboards and screens and turn the big printer into a v2.

Egosumumbravir, 2023-11-06:
> I'm putting mine on an ender 3 pro with the 4.2.7 board and 400x400 build plate. It says mriscoc isn't compatible with the pro.

The changes required to get the MicroProbe working should apply to regular Marlin too, I don't think there's anything special about it.

Basically: the BLT port on the 422/427 boards does not have sufficient pullup on the sense pin to work correctly with a v2 MicroProbe - nor does it use BLT protocols.
However, by telling Marlin it's a clicky probe and using the original Z endstop port which does have sufficient pullup, the system has worked fine for hundreds of cycles for me.

A v427 board should be fine running MRiscoC, but I'm not sure about the display - that might be the issue with compatibility,

PandaMiann, 2023-11-19:
> UPDATE: I have the Microprobe v2 working as expected with a custom compiled MRiscoC Professional Firmware. https://github.com/mriscoc/Ender3V2S1

Hey! Could you please provide your edited firmware or write down the specific changes you've made? I also have Ender 3V2 4.2.2 board trying to run the v2 microprobe, however I've never played with the firmware and I'm really scared that I mess something up xd. I really like the MRiscoC firmware so that would be exactly what I need! Thanks!

Egosumumbravir, 2023-11-22:
> Hey! Could you please provide your edited firmware or write down the specific changes you've made?

Now I've been running it for a while and am sure there's no strange behaviours I reckon it's safe. Noting above that the cable needs to be split between the BLT port for probe deployment and the Z-stop for actual detection of course.

Starting with a custom MRiscoC configuration.h with BLT enabled so that UBL is enabled.
Lines and changes
1179: #define Z_MIN_PROBE_ENDSTOP_HIT_STATE LOW
Default is HIGH, needs to be LOW for a v2 MicroProbe

1340: #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN  // Probe connected to BLTouch port
Miguels BLT disables this, we need to REENABLE it to use the old Z-MIN endstop port (PA7)

1343: #define USE_PROBE_FOR_Z_HOMING
Already enabled by BLT config

1358: #define Z_MIN_PROBE_PIN PA7
Defaults to disabled, specify the non-BLTport - PA7 is the traditional 422/427 z-min endstop port

1378: #define FIX_MOUNTED_PROBE
Usually disabled for BLTOUCH, needs to be enabled

1400: //#define BLTOUCH  // 3D/CR/BLTouch version
Microprobe is is NOT a BLTOUCH clone, must disable

1604: #define PROBE_ENABLE_DISABLE
Needs to be enabled for the microprobe v2

1606: #define PROBE_ENABLE_PIN PB0   // Override the default pin here
Microprobe uses the default BLT port to deploy. PB0 is that pin on the 422/427 boards.

good luck

PandaMiann, 2023-11-24:
> Now I've been running it for a while and am sure there's no strange behaviours I reckon it's safe. Noting above that the cable needs to be split between the BLT port for probe deployment and the [...]

Hey! Thank you so much for your time and effort you are a lifesaver! My microprobe shall arrive any day now, so I'll leave my feedback here after my attempt. Once again, thank you!

PandaMiann, 2023-11-27:
Okay so I've used your configuration and everything works great! I would just like to add that when you set line 1179 to LOW, then you also need to set line 1165 to LOW otherwise the Auto Build Marlin doesn't let you compile the code. The physical installation itself was quite a pain in the ass. The cables were too short and have dupont connectors, so they don't fit very well. I used hot glue to make sure that they don't disconnect during printing. Overall the probe works great, and saves me a lot of time.
@Egosumumbravir  Thank you so much!

Egosumumbravir, 2023-12-02:
> I would just like to add that when you set line 1179 to LOW, then you also need to set line 1165 to LOW otherwise the Auto Build Marlin doesn't let you compile the code.

Cheers, I missed that edit!

For future reference
1179: #define Z_MIN_ENDSTOP_HIT_STATE LOW //Default is HIGH, needs to be LOW for a BQMPv2

Lykeosovandara, 2023-12-21:
I can't build the filmware 
[image]

Lykeosovandara, 2023-12-22:
> Cheers, I missed that edit!
>
> For future reference 1179: #define Z_MIN_ENDSTOP_HIT_STATE LOW //Default is HIGH, needs to be LOW for a BQMPv2

Can I have your hard connection?

delong369, 2023-12-22:
I did not customize my own firmware I had Miguel Risco do my custom
firmware for me for the BQ microprobe. I paid him $5 to do it. You will
customize any kind of firmware you want

Mogi-R, 2024-02-20:
Could you describe specifically which release of MRiscoC you used (according to the line number I suspect `20230805`) and how you built it?
I keep getting errors using Auto Build Marling in VScode, one is about `Z_SAFE_HOMING` which is easy to solve by commenting it out on the config file, and others are changing with every release, with the one I stated is `Marlin\src\lcd\e3v2\proui\dwin.cpp:2880:63: error: 'probeTest' was not declared in this scope`

and also which way did you connect the probe? using all 5 wires in the bltouch port or one of them in the z axis port

Mogi-R, 2024-02-23:
Thanks for the elaborate answer!
But I still cannot make it work, when I use the default config from the release with the changes @Egosumumbravir suggested the build still fails (tried the last 4 releases with different errors, most related to DWIN)
When I use the config made using mriscoc's Special_Configuration (`ender3V2, 422, AutoLev or BLT (both didn't work) with UBL, correct display the thermistor and HomeOffs, IS, LA, MPC`) and the suggested changes it builds and I can install it correctly, but when I try to `Auto home` the printer keeps lowering without the probe triggering it to stop.

I'm pretty sure I'm doing something wrong with the configuration and the firmware, unless you think there is something wrong with my probe itself..

Any suggestions?

Egosumumbravir, 2024-02-24:
> Any suggestions?

Start with the basics: Advanced -> End Stops Diag.
My probe reports "z-min: TRIGGERED" in the default retracted state and "open" when I pull it down. It happily cycles between them as I manually move the probe.

Mogi-R, 2024-02-24:
My probe reports `OPEN` in both situations
the suspects are firmware (maybe there is something wrong with the configuration), wiring (but I wired the probe them same as suggested) or my probe is defected.
Is there a way to isolate each suspect to make sure it works ok?

Egosumumbravir, 2024-02-24:
> Is there a way to isolate each suspect to make sure it works ok?

You can test the physical action of the sensor by plugging the old switch back in and simply clicking it. This might indicate flawed hardware or you've missed

1340: #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN // Probe connected to BLTouch port Miguels BLT disables this, we need to REENABLE it to use the old Z-MIN endstop port (PA7)
or
1358: #define Z_MIN_PROBE_PIN PA7 Defaults to disabled, specify the non-BLTport - PA7 is the traditional 422/427 z-min endstop port

Does your probe do it's bootup routine when you power the machine on - cycles deploy and retract three (I think it's three?) times.

Mogi-R, 2024-02-24:
> You can test the physical action of the sensor by plugging the old switch back in and simply clicking it. This might indicate flawed hardware or you've missed
The old switch works just fine in the diagnostics (the opposite obviously, because of the firmware configurations)

> Does your probe do it's bootup routine when you power the machine on - cycles deploy and retract three (I think it's three?) times.
It doesn't

and I did use these configurations

Mogi-R, 2024-02-24:
Finally figured it out! turns out the firmware I built worked ok, but when I tested the extension cables I used for the probe with a multi meter were defected..

Just for future references summering what I used to make it work:

- Used the special configuration interface with Ender3V2, 422, BLT w UBL, DWIN, T1, Homeoffs + IS + LA + MPC
- changed the configurations according to the comments here
- used the latest release of mriscoc (20240122) and built the firware

Thanks everyone for the help

hello42world, 2024-06-15:
Thanks a lot. This helped me make the probe work with 4.2.2 mainboard based Ender 3. Below is the config patch that I ended up with. 
[micro_probe_e422.patch.txt](https://github.com/user-attachments/files/15847725/micro_probe_e422.patch.txt)

Built off of the Marlin bugfix-2.1.x branch.

ltxtxl, 2024-07-01:
@hello42world Hi are you able to send me the file, i'm totally new to coding and have no item how to compile

hello42world, 2024-07-07:
@ltxtxl What file? The configuration patch is attached to my message. You don't need to code anything if you just need to build Marlin. 
https://marlinfw.org/docs/configuration/configuration.html - Refer to Marlin docs for details

atprieto, 2024-08-08:
> Muchas gracias. Esto me ayudó a hacer que la sonda funcione con la placa base Ender 3 4.2.2. A continuación se muestra el parche de configuración que terminé usando: [micro_probe_e422.patch.txt](https://github.com/user-attachments/files/15847725/micro_probe_e422.patch.txt)
> 
> Desarrollado a partir de la rama Marlin bugfix-2.1.x.

hola amigos. 
llevo dias intentando configurar la sonda, hasta llegar aquí, no tengo un conocimiento tan profundo sobre el manejo del codigo de la impresora, con lo que mi pregunta es sencilla.

entiendo que el parche que ha colgado [hello42world] he de introducirlo en mi tarjeta SD y ponerselo a mi ender 3 Max?

DvaKompota, 2024-10-07:
## WORKING SOLUTION:
My CR Touch died so I decided to change it to MicroProbe V2 (without much research 😕 )
My setup:
> Ender 3 V2 (MB Creality v4.2.2) running Klipper:
> Version: v0.12.0-286-g81de9a861
> OS: Raspbian GNU/Linux 11
> Distro: MainsailOS 1.2.1

Had all the mentioned issues + some of my own: gantry going the wrong way, probe not detecting the touch, probe always open or always triggered, etc.

I reused the CR Touch wire (with idiotic color-coding) as it was already in place.
Had to do the probe re-wiring to Z-Min, like described [here](https://github.com/bigtreetech/MicroProbe/issues/14#issuecomment-1955080634). 
No soldering was needed — just pushed out of the 5-pin connector two sensor contacts and put them to the former Z-stop connector which was scary, but worked without burning the board.
BUT BE SURE TO DOUBLE CHECK THE WIRING SCHEME BEFORE TURNING IT ON!

Here's the config that finally worked for me (my old CR Touch config below is for reference):
```
[gcode_macro Probe_Deploy]
gcode:
    SET_PIN PIN=probe_enable VALUE=1

[gcode_macro Probe_Stow]
gcode:
    SET_PIN PIN=probe_enable VALUE=0

[output_pin probe_enable]
pin: PB0 # Change this to the Servo/control pin number for your board.
value: 0

[probe]
pin: !PA7 # Change this to the Probe/Z- pin number for your board. Remove the ! for V1 probes.
deactivate_on_each_sample: False
x_offset: -42
y_offset: -10
#z_offset: 0
speed: 5.0
samples: 2
#sample_retract_dist: 2.0
samples_tolerance: 0.05
samples_tolerance_retries: 3
activate_gcode:
    Probe_Deploy
    G4 P1000 # allow time for probe to deploy before homing Z
deactivate_gcode:
    Probe_Stow

# [bltouch]
# sensor_pin: ^PB1
# control_pin: PB0
# x_offset: -42
# y_offset: -10
## z_offset: 0
```

Hope it helps.

Zodi4co, 2024-11-29:
Olá, cheguei até aqui com o mesmo problema, minha impressora é uma Ender 3, 4.2.2, e não consigo fazer o microprobe funcionar, ja instalei vários firmware e nenhum funciona, alguém pode compartilhar o código. Vi várias dicas de alterações nos códigos, mas eu realmente não sei fazer !