bigtreetech/MicroProbe #18: Microprobe doesn't work with SKR Pico

PythonesqueSpam, 2023-12-15:
I've made changes to Klipper on my SKR Pico according to instructions herein and checked the wiring and the probe deploys and stows but never shows as triggered. The probe deploys using the Probe_Delpoy macro but doesn't retract when touched  lightly with a fingernail.  On attempting to home the probe does nothing and the carriage crashes into the bed.

Al previous efforts with the BL Touch, CR Touch, 3D Touch and IR Differential probes all failed. So I switched to the Microprobe as it is from the same manufacturer as my Pico, in the false hope that they should work better together. 

After weeks of messing with hardware from this manufacturer I'm not inclined to do carry on doing testing they should have done and will just return both the Microprobe and Pico as unfit for purpose, if this issue is not resolved quickly.

gfunkbus76, 2023-12-26:
No luck?

nhusby, 2023-12-28:
I got it working with my Pico, it homes and meshes, but won't retract.  

```
[output_pin probe_enable]
pin: gpio29 # The control IO pin 
value: 0 # Probe default retracted

# Probe deploy command
[gcode_macro Probe_Deploy] 
gcode:
  SET_PIN PIN=probe_enable VALUE=1

# Probe stow command
[gcode_macro Probe_Stow] 
gcode:
  SET_PIN PIN=probe_enable VALUE=0 

[probe]
pin: ^!gpio22 # The detection IO, ^ means high-level triggered, V2 should be ^! means low-level triggered.
deactivate_on_each_sample: False
x_offset: -33.0 # Actual installed offset of MicroProbe 
y_offset: -9.0 # Actual installed offset of MicroProbe 
#z_offset: 0.0 # Actual installed offset of MicroProbe 
speed: 10.0
activate_gcode:
  Probe_Deploy
  G4 P500 # Allow 500 milliseconds for the probe to deploy 
deactivate_gcode:
  Probe_Stow
  
```

spool3dcanada, 2024-01-16:
Just ran into this today.  You need to use the z- stop connection for the Micro Probe connection with the black and white wires. In printer.cfg:

[probe]
pin: ^gpio25

NunoFilipePereira, 2024-01-31:
Can you better explain this solution? the physical connection and then in the file. Thanks

jacobg1496, 2024-03-03:
Also having this same issue and no luck in resolving it. Have went so far as to order a second probe to make sure I did not damage the probe somehow.

JdeV987, 2024-09-07:
This is I got mine to work:
All of below must be in  printer.cfg

`
[output_pin probe_enable]
pin: gpio29                                 # The control IO pin 
value: 0                                    # Probe default retracted

;# Probe deploy command
[gcode_macro Probe_Deploy] 
gcode:
        SET_PIN PIN=probe_enable VALUE=1

;# Probe stow command
[gcode_macro Probe_Stow] 
gcode:
      SET_PIN PIN=probe_enable VALUE=0 

[probe]
;# You need to use the z- stop connection for the MicroProbe connection with the black and white wires. 

pin: ^!gpio25                         # The detection IO
deactivate_on_each_sample: False
x_offset: 0.0                         # Actual installed offset of MicroProbe 
y_offset: 0.0                         # Actual installed offset of MicroProbe 
z_offset: 0.0                         # Actual installed offset of MicroProbe 
speed: 10.0

activate_gcode:
       Probe_Deploy
       G4 P500                           # Allow 500 milliseconds for the probe to deploy 
deactivate_gcode:
       Probe_Stow

[stepper_z]
;# ? to be replaced by your own values
step_pin: gpio19
dir_pin:  !gpio28                    # You may or may not need to invert the signal with a !  (depending on your Z-motor direction.)
enable_pin: !gpio2
endstop_pin: probe:z_virtual_endstop

microsteps: ?
rotation_distance: ?       
position_min: ?
position_max:  ?
homing_retract_dist: ?
homing_retract_speed: ?
homing_speed: ?
second_homing_speed: ?
`

[image]