bigtreetech/MicroProbe #19: Compatibilty with EBB SB2209 / RP2040 CAN board

kloboucek, 2024-01-10:
Hello @bigtreetech team,

I've been trying to get the microprobe v2 work with this board and I can't get it to sense when it is triggered. I'm running Klipper on Voron Trident with SB toolhead and the RP2040 based CAN board - previously with Omron proximity probe which worked fine, but the microprobe is either always triggered and the printer won't home or it is never triggered and it's crashing into the bed.

My probe config is:

```
[output_pin probe_enable]
pin: EBBCan:gpio22
value: 0

[gcode_macro Probe_Deploy]
gcode:
 SET_PIN PIN=probe_enable VALUE=1

[gcode_macro Probe_Stow]
gcode:
 SET_PIN PIN=probe_enable VALUE=0

[probe]
pin: ^!EBBCan:gpio21 ## For V1 version, set to ^pin for high-level trigger; for V2 version, set to ^!pin for low-level trigger.
deactivate_on_each_sample: False
x_offset: 0.0 # Actual offset of the MicroProbe installation
y_offset: 25.0 # Actual offset of the MicroProbe installation
z_offset: 0.0 # Actual offset of the MicroProbe installation
speed: 5.0
activate_gcode:
 Probe_Deploy
 G4 P500 # Allow 500 milliseconds for the probe to deploy
deactivate_gcode:
 Probe_Stow
```

With this configuration it is always triggered even when the pin is pushed out. If I invert the sense pin, then it's always open, but won't change to triggered upon surface contact or manual pin push.

I'm reading in other posts that the BLT connector may not have sufficient pull up on sense pin to work with this probe and that the only option known to work is to use, for example, the original Z endstop pin on the main MCU. I'm considering testing this option out, but if anyone at @bigtreetech knows of a solution that works solely with the EBB SB2209/RP2040, it would be much appreciated.

Thanks...
Michal

kloboucek, 2024-01-10:
So, I have found a workaround that might be specific ty mine, but likely also many other Tridents or 2.4r2's out there. 

First I tried using the original Z-endstop pin as the sensing pin, but the probe was behaving exactly the same as before when connected to the BLT port on SB2209/RP2040. Meaning no change of state no matter if the probe pin was in our out of the probe body.

I'm using Octopus 1.1 as my main MCU and this board also has a BLT port on it, so I thought why not giving it a try and I connected the sensing pin to ^!PB6 and hey, it started to work like a charm. So now the CAN board controls the pin movement and BLT control pin on the Octopus is sensing the state. I guess I can handle one extra pair of wires going down from the toolhead via the umbilical along with the CAN cable... but it's a bit sad, it would have been better if it worked right out of the box with the BLT port on the CAN board. I did try swaping the wires in the connector so that the sense wire and ground go to control pin and ground on the CAN board and control wire and its ground go to the sense pin and ground on the CAN board, but no luck there, it only works when I use the BLT on the Octopus for sensing the change of state.

This issue can be closed, but I'd very much appreciate @bigtreetech to provide an official confirmation whether it is possible to make it work somehow or that these two components were not intended to work together. I presume the original BLT probe would work fine with the CAN board.

Anyway, thanks and happy printing :-)

Gardosen, 2024-02-24:
I think i have the same issue with an SKR 3.

thise seems to be a common issue when combining the EBB2B2209 RP2040 with a microprobe.

kloboucek, 2024-03-08:
So, it's kinda irrelevant what the main MCU is, the probe just doesn't play along well with the RP2040 controlled EBB board. I have another one paired with EBB36 and it works flawlessly there. And I'm about to install one more on a printer without CAN directly plugged to Octopus Pro board - I've tested it with one of those I had already and had no issues there with the sensing pin.

IggyTheWolf, 2024-06-23:
I'm having the same issue with the SB2209. I can get the microprobe to deploy and stow, but it's showing as constantly triggered. I have the same configuration as above.

rafaelrabaco, 2024-07-12:
Did anyone make it work? Has anyone tried installing the pull up resistor?
[image]

rafaelrabaco, 2024-07-12:
If you connect it like this below, you already have the 10k resistor on the board
[image]
[image]

IggyTheWolf, 2024-07-12:
> Did anyone make it work? Has anyone tried installing the pull up resistor?
> 
> [image]
> 
> 

I have tried everything and it doesn't work configured like that. 

The solution is the picture posted above.

rafaelrabaco, 2024-07-13:
> I have tried everything and it doesn't work configured like that.
>
> The solution is the picture posted above.

I'll test method one, let's see if it works

DVSVIDEO, 2024-09-15:
Az az igazán bosszantó, hogy sehol a reklám anyagban és az eladó sem a gyártó oldalán nincs megemlítve, hogy nem kompatibilis a 2209 CAN kártyával...
Biqu and Bigtreetech are the same manufacturer! And then they make a device that is not compatible with a device whose connector is supposed to accept this probe...
The problem is that I order it, wait half a day to receive it, install it, configure it and it doesn't work. Then I look for it...and then I find out that it doesn't work for others...how did the manufacturer test it????
Why no warning?
*** Translated with www.DeepL.com/Translator (free version) ***

bigtreetech, 2024-09-15:
> What's really annoying is that nowhere in the promotional material or on the vendor's site is it mentioned that it is not compatible with the 2209 CAN card... Biqu and Bigtreetech are the same manufacturer! And then they make a device that is not compatible with a device whose connector is supposed to accept this probe... The problem is that I order it, wait to receive it....and half a day install it, configure it and it doesn't work. Then I look for it...and then I find out that it doesn't work for others...how did the manufacturer test it???? Why no warning? *** Translated with [www.DeepL.com/Translator](http://www.DeepL.com/Translator) (free version) ***

It is documented in the manual for the SB2209 RP2040:
[image]

closed 2024-09-15 by bigtreetech