bigtreetech/MicroProbe #20: Stylus does not rise after removing the table card on the MKS Robin Nano v3.2

mruttc, 2024-01-14:
Good afternoon, everyone!
Today I became a happy owner of Microprobe v2.
I have encountered an incredibly terrible problem. On my MKS Robin Nano V3.2 board.

The microprobe stylus does not always go in after building the map before printing. Therefore, the stylus itself moves around on the table.

I also noticed that the PROBE_UP command does not work, but a faint click is heard (the stylus remains at the bottom). Sometimes everything works fine.

https://youtu.be/xa_sol0xll0

```
[output_pin probe_enable]
pin: PA8
value: 0

[gcode_macro PROBE_DOWN]
gcode:
    SET_PIN PIN=probe_enable VALUE=1

[gcode_macro PROBE_UP]
gcode:
    SET_PIN PIN=probe_enable VALUE=0

[probe]
pin: ^!PC4
deactivate_on_each_sample: false
x_offset: 41
y_offset: 5
#z_offset: 0.618
samples: 3
samples_tolerance: 0.01
samples_tolerance_retries: 3
activate_gcode:
    PROBE_DOWN
    G4 P500
deactivate_gcode:
    PROBE_UP

[bed_mesh]
speed: 100
horizontal_move_z: 10
# mesh_min: 41, 15
# mesh_max: 320, 320
mesh_min: 41, 15
mesh_max: 290, 320
probe_count: 9, 9
mesh_pps = 3, 3
algorithm: bicubic

[safe_z_home]
home_xy_position: 128, 160
speed: 50
z_hop: 10
z_hop_speed: 10
move_to_previous: false
```

mruttc, 2024-01-14:
Here is the pinout of the board used.
[image]

guot0202-source, 2025-11-27:
I encountered issues with the microprobe and MKS ROBIN NANO V3.0 mainboard.
Is there a configuration error? I believe not, as the PROBE_DOWN function operates correctly, only PROBE_UP fails to work. Accuracy testing has not yet been conducted.
Doesn't work.
Only Deploy, don't Stow.

[output_pin probe_enable]
pin: PA8   #For MKS Robin Nano v3.0
value: 0    # Initial state (0 = Stow)

# Probe deployment command
[gcode_macro Probe_Deploy]
gcode: SET_PIN PIN=probe_enable VALUE=1

# Probe retraction command
[gcode_macro Probe_Stow]
gcode: SET_PIN PIN=probe_enable VALUE=0

[probe]
pin: ^PC8        #For MKS Robin Nano V3.0 
activate_gcode: Probe_Deploy G4 P500 #allow 500 milliseconds for probe deployment
deactivate_gcode: Probe_Stow  
x_offset: 0
y_offset: -22
#z_offset: -1.45
speed: 5
samples: 3
samples_tolerance: 0.01
samples_tolerance_retries: 6