bigtreetech/MicroProbe #22: Endstop z still triggered after retract

MrGray942, 2024-02-04:
Manta M8P with HermitCrap V1 was working fine with microprobe V1.2. After the pin get damaged and replaced with microprobe V2 it keeps given the some error, the set up is okay and regardless any changes applied I still get the same error. brought the second probe and have the same situation. Is any solution for it. I have seen  lotos of the some of similiar cases and no answer from bigtreetech

javcab2003, 2024-02-10:
same her I will try to connect a CR-Touch that I have and test

javcab2003, 2024-02-11:
Dear People, I need your help.

I have installed the Biqu Micro Probe in a new Manta EZE3 with driver TMC5160 RGB ( I think also are PRO)

I have a double Z axis setup using the frame of my Ender 3.

After hours of research and reading similar issues as mine, I am not sure now where is coming the problem.

In the start of the journey to find the solution to this error: " Endstop Z still triggered after retract",  and after trying with different options with changing position of the cables for the Z endstop following the instructions from BTT, changing the the printer.cfg settings following carefully the official instructions to make work the probe, I came to the conclusion that I am desperate.

Right now the Y,Z go homing but Z move a little bit down and then stops, is not homing. The probe is deploying in the air because Z is not homing. After hours of testing different settings with cable and .cfg,  the system right now is not displaying anymore the error " Endstop Z still triggered after retract" but I can hear the Z motors in stall mode, and the Z motors go very hot  if I leave it some minutes in this way.
You will see in the .cfg file the pins are correctly set for the mainboard EZE3. I have checked 100 times the pin position and even play it with different pins positions using the Z endstop changing the pin in the settings, but the only change is that instead of going down the Z, is going up. 

Maybe some of you have some idea how to set my mistake or probably I have a defect probe. I even tried with a CR touch that I have here, but didn't work correctly and I think I need to change the cable for the CR touch because I was using the original cable from the Biqu Probe.

My Printer.cfg here ( At the end is the MicroProbe.cfg with the settings):

[include mainsail.cfg]

[include MicroProbe.cfg]

[stepper_x]
step_pin: PA14
dir_pin: PA10
enable_pin: !PA13
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC4
position_endstop: 20
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PC8
dir_pin: PA15
enable_pin: !PC14
microsteps: 16
rotation_distance: 40
endstop_pin: ^PB0
position_endstop: 45
position_max: 245
homing_speed: 50

[stepper_z]
step_pin: !PD2
dir_pin: PD4
enable_pin: !PD3
microsteps: 16
rotation_distance: 40
gear_ratio: 80:16
full_steps_per_rotation: 200 #200 for 1.8 degree, 400 for 0.9 degree
endstop_pin: probe:z_virtual_endstop #^PC6
#position_endstop: 0
position_max: 250
position_min: -8
homing_speed: 25
homing_retract_dist: 0.0

[stepper_z1]
step_pin: PB7
dir_pin: !PB6
enable_pin: !PB4
microsteps: 16
rotation_distance: 40
gear_ratio: 80:16
full_steps_per_rotation: 200 #200 for 1.8 degree, 400 for 0.9 degree
#endstop_pin: probe:z_virtual_endstop

[z_tilt]
z_positions:
[... 220 lines cut ...]
value: 0

[probe]
pin: ^!PA6         # Change this to the Probe/Z- pin number for your board. Remove the ! for V1 probes.
deactivate_on_each_sample: False
x_offset: -39.4                    # Change the probe offsets to meet your machine requirements.
y_offset: -1.5
z_offset: 0
speed: 5.0
samples: 2
sample_retract_dist: 2.0
samples_tolerance: 0.05
samples_tolerance_retries: 3
activate_gcode:
    Probe_Deploy
    G4 P500
      #allow time for probe to deploy before homing Z
deactivate_gcode:
    Probe_Stow

[safe_z_home]
home_xy_position: 145.6, 90   # Change this is the coordinate for the center of the bed
speed: 50.0
z_hop: 10
z_hop_speed: 15.0

Kruworkz, 2024-02-14:
same issue started today. the z will drop when homing then get endstop z still triggered after retract.  anyone have a fix?

jokhacker-maj, 2024-02-22:
Hi guys i am having the same problem with exact same harware. Just watched https://www.youtube.com/watch?v=q6ffmH99hRA this video and now i can move the pin down. But when i try to home the machine the pin goes down, head moves 2 cm up and than pin goes up and error shows.

javcab2003, 2024-02-22:
share with me your printer.cfg I have resolved the issue. I just found the the way to setup the dir_pin with or without the ! symbol. But I need to see first your code and you need to explain me your printer, do you have double Z, ender 3? just to know.

jokhacker-maj, 2024-02-23:
Thanks man i just resolved it 1 hour after original post. It is exactly what you said. I forgot to put ! in front of the pin. Thaks, happy printing.

Aurvandill, 2024-03-21:
If you replaced the MicroProbe V1 with a V2 you need to invert the z Endstop pin
on my manta m4p it's the following
```yaml
pin: ^!PC14 # ^PC14 for V1 Probe
```

Kruworkz, 2024-03-21:
> [probe]pin: ^!PC14 # ^PC14 for V1 Probe

It's a v2 on a manta m8p

Kruworkz, 2024-03-22:
This is my probe printer config file section for probe.

cosmindv, 2024-05-06:
My problem with the biqu microprobe is a hardware one, i get the same error "Endstop z still triggered after retract" the pcb is loose and moving outside of the body of the probe, i push it back in with 1 cm and the error disappears. i think the probe pin hit the printed part too much and the pcb is loose because of that.

3d-design-fun, 2024-07-25:
I had the same problem. Apparently the microprobe is very sensitive due to the short travel of the pin. If you set the: deactivate_on_each_sample: True then it works without issue. It may not be as speedy, but it is reliable.