bigtreetech/MicroProbe #25: Modern MRiscoC/Marlin bugfix branches versus a SKR Mini E3 v3 & Microprobe v2

Egosumumbravir, 2024-02-25:
Starting this thread as I work through getting an up-to-date bugfix Marlin implementation running on my not-very-Ender 3v2.

After getting the probe successfully operational on my Creality 4.2.2 board ( https://github.com/bigtreetech/MicroProbe/issues/14 ), I've hit a roadblock where Linear Advance is sending the extruder stepper catatonic mid-print. LA is so vital to exceptional print quality I've decided to upgrade the control board to a SKR Mini E3 V3.0 to bypass the issue entirely ... and I don't feel like taking a soldering iron to my Creality board ... yet anyway.

Using the latest source from MRiscoC Professional Firmware (20240122) https://github.com/mriscoc/Ender3V2S1

The first real issue I encountered when working my way through the manual pages which seem relevant to my setup (pages 20, 21 & 22) is this: Z_MIN_PROBE_ENDSTOP_INVERTING

It does not appear to exist in the 2.1.x bugfix branch of Marlin that MRiscoC is based upon.

For my 422 configuration, I've used "Z_MIN_ENDSTOP_HIT_STATE" & "Z_MIN_PROBE_ENDSTOP_HIT_STATE" to work around the issue there. Guess I'll have to see if it works here. Still have a couple of days before the postie rocks up with my new E3 board...

bmourit, 2024-03-05:
This is do to the fact that you can now simply enable support for the MicroProbe (V1/V2) directing in the config support and Marlin will handle the correct inversion.

Egosumumbravir, 2024-03-05:
Which will be great for the next MRiscoC release built upon this.
For the moment, guess I'll have to get experimental.

jftaylorMn, 2024-03-09:
@bmourit ,  are you saying that the latest marlin configuration.h file has direct support for MIcroprobe V2? can you elaborate?  I am trying to build firmware based on marlin for my ender3 pro with 4.2.2 board.  After fiddling for a few days using suggestions form a reddit thread, I have the expected menu items which generally work as expected- except that Move Axis fails after AutoHome.    I think the microprobe (connected to the z axis limit) is working as intended (since autohome doesn't crash into the bed).

The only response I have from BigTreeTech support via email is "read the manual", and "we support our boards".

bmourit, 2024-03-09:
Yes @jftaylorMn The latest Bugfix has the option in the Configuration.h  file to select Biqu MicroProve alone with the version.  It also states  the other things that you will need to enable directly there in the comments.

bmourit, 2024-03-09:
Just a note, I have the Ender-3 with the 4.2.2 board as well. For this one, I ended up having to move the z-endstop pins on the MicroProbe and use them directly on the Z endstop pins from the endstop I remove. Then also making sure the z probe is marked as using the endstop pins. This is due to the pull-up requirements, which the ender 3 board seems to not have on the bltouch slot pins.

jftaylorMn, 2024-03-09:
@bmourit Adding the microprobe has caused me to venture into updating firmware on the ender3.  I have done a lot of microcontroller firmware, but not specifically for a 3d printer.  It's my first exposure to the internals of Marlin. I did see many references to using the z axis limit connection instead of the pins on the bl touch plug.  The reddit reference that looked pretty complete is [here](https://www.reddit.com/r/BIGTREETECH/comments/169jbq0/biqu_microprobe_marlin_setup/). 

I started with the example creality ender3 4.2.2 configuration example from the marlin github and got that to compile (and flash).  I have since compared that configuration.h file to the default that comes with a marlin 2.1.2.2 and see the new section dealing with the microprobe,   I will define BIQU_MICROPROBE_V2, but am not sure what that does "under the covers".. It likely makes some of the other changes unnecessary.   My latest file already has PROBE_ENABLE_DISABLE defined and #define PROBE_ENABLE_PIN PB0.  Beyond what was suggested in the reddit link mentioned, I have also enabled the bed leveling menus and maybe a few other settings to get a clean compile.  I'm guessing it's one of those other changes that is causing the move axis menu options to fail after auto home.  That's just a guess for now.

I sense some quality time with the notepad++ compare plugin and/or windiff to understand better what changes to try.

If your configuration works, would you be willing to post it here for reference?  I'm sure that I'm not the only one trying to get this to work.  Who knows, maybe BigTreeTech would add it to this github and be able to say more than "read the manual".  No warranty expressed or implied, of course.  If you would rather not post it publicly, feel free to email it.

Egosumumbravir, 2024-03-09:
> I am trying to build firmware based on marlin for my ender3 pro with 4.2.2 board.

@jftaylorMn right here buddy: https://github.com/bigtreetech/MicroProbe/issues/14

jftaylorMn, 2024-03-10:
@Egosumumbravir -  I appreciate your link to [issue 14](https://github.com/bigtreetech/MicroProbe/issues/14), however that solution doesn't make use of the new configuration.h with #defines for the Microprobe_V1 or V2 that @bmourit mentions. While I have not actually built the firmware based on the configuration in https://github.com/mriscoc , it appears that the code is for the improved graphics display of the V2, not ender-3 Pro.  There could be other changes that aren't going to work for the ender-3 Pro. These configuration files are very large (3700 lines),  and they change over time (depending on marlin release).  Rather than summarizing in text how to update the file (without defining the baseline file), it would be far simpler to just present a configuration.h that is known to support the microprobe on a stock ender3-pro with a 4.2.2 board.  I'm expecting that the new #defines for the microprobe will reduce the number of other changes needed.  Maybe not?  we'll see.

Egosumumbravir, 2024-03-10:
Yep, for the moment I'm seeing about modding the old version while I wait for MRiscoC to update his.
Not a lot of difference between a Pro and a v2 mechanically, although depending on the v422 board revision you might have different stepper controllers.
Yep, a premodified configuration.h would be great. Aren't DIY printers the best fun!

bmourit, 2024-03-10:
The new config just takes makes sure the proper inversions are done based on the version of MicroProbe.
I'll attach my config files.
[Archive.zip](https://github.com/bigtreetech/MicroProbe/files/14549861/Archive.zip)

reyesv351, 2024-03-10:
> The new config just takes makes sure the proper inversions are done based on the version of MicroProbe. Beyond that, it mainly just replaces the the need for enabling the FIX_MOUNTED_PROBE define. I'll attach my config files. [Archive.zip](https://github.com/bigtreetech/MicroProbe/files/14549861/Archive.zip)

this is pretty awesome they finally fixed it but I'm assuming this is the BugFix 2.1.x version of Marlin and not the stable version yet?

jftaylorMn, 2024-03-10:
@reyesv351 ,  yes this is currently only in the bugfix branch.  I found the [pull request](https://github.com/MarlinFirmware/Marlin/commit/204de723f1c08f83e281f0629b60822b334097ed) that covers all the changes made.  If you want, you could add only these updates to your environment -  that might reduce your concern for what might be changed.  There could be prerequisites also in the bigfix branch that you would need to bring along.   

@bmourit -  thanks for including your configuration files.  In addition to probe specific items, I see some changes to default feed rates, acceleration, and maybe some PID optimization.  Would you recommend taking those or reverting to the default values from the creality example configs?

bmourit, 2024-03-14:
@jftaylorMn 
The PID I usually run from the LCD screen and then save to eeprom to make sure the calibration is correct. If you aren't having issues, you can ignore my changes. I do the same with acceleration and fr, so those numbers may not actually be the ones I later configured after tuning, but just the starting point. I change them based on filament used and how complex the object I am printing is, so it is highly user discretionary.

jftaylorMn, 2024-03-14:
@bmourit - PID tuning is something I studied in engineering school decades ago and have never had a reason to use- possibly until now.  Truth is that I am on my first spool of filament and not straying too far from "defaults" provided by Creality and Prusa Slicer.  I'm OK with the results as long as I get good bed adhesion.  I suspect my prints could be done faster if I paid attention to speeds, acceleration, etc. I wanted to put any bed level doubts in the rear view mirror first.  For now, I'll enable the advanced menu for PID tuning, and stick with the values from the configuration.h file for the ender3 pro.   Thanks again for your help!

bmourit, 2024-03-14:
No prob! The default values come from someone running the tuning on the same machine, so you should be good. I can say, once I got this MicroProbe working well, my bed adhesion has been very nice so far. There are still some types of filaments I use a glue stick with though to help with removal, but I have to say, I'm loving this MicroProbe now.

ViktorDomzoro01081981, 2024-03-17:
> Это связано с тем, что Marlin теперь имеет встроенную поддержку этих зондов. Просто включите поддержку MicroProbe (V1/V2) непосредственно в Configuration.h, и Marlin выполнит правильную инверсию.

Не работает! Я проверял

jftaylorMn, 2024-03-17:
@ViktorDomzoro01081981 - After some manipulation of the files that @bmourit provided (above), I have this working now on my ender 3 pro (three microprobe wires connected to the probe socket and two replacing the z axis limit switch). Be sure that G and V are connected to the right pins on your board.  The Microprobe should extend/retract three times at startup. 

Unless a new official release of marlin is out, you will need the 2.1.x (bugfix) branch code.  The configuration and configuration_adv.h files are specific to marlin version, so you may need to do some "merge conflict resolution" to create the files if you have older files with configuration data you care to keep.  The "stock" printer/board example configuration files provided in the marlin "configurations" github should compile and flash successfully.  There may be features (like bed leveling) that are not enabled by those files.  That's where line by line comparing with @bmourit's files may help out.

bmourit, 2024-03-17:
I am glad you got this working!! The probe itself is quite good. Enjoy!