bigtreetech/MicroProbe #26: SKR Pico and Microprobe Not Working

jacobg1496, 2024-03-03:
As described in other issues, I am having trouble getting the BTT Microprobe to work with the Pico. I have everything wired up correctly using the servos and probe ports on the board, have removed the jumped for sensorless z homing, and have made sure to add what I thought were the necessary values in to my printer.cfg and btt_microprobe.cfg files. 

However, every time I try to run a home or a calibrate probe, it will deploy the probe and the bed will just continue going up crashing in to the nozzle. 

probe_stow and probe_deploy both seem to work. 

I'm at my wits end here honestly. I have tried 2 separate probes that I purchased to rule out the chance of me getting a dud probe or wiring it incorrectly and damaging it myself. 

If anyone is able to help me out I would be super appreciative. 

Here is my printer.cfg as it is right now.

`[include fluidd.cfg]
[virtual_sdcard]
[include btt_microprobe.cfg]
#path: /home/klipper/printer_data/gcodes
#on_error_gcode: CANCEL_PRINT

#####################################################################
#      General Settings
#####################################################################

[mcu]
serial: /dev/ttyAMA0
restart_method: command

[printer]
kinematics: corexy
max_velocity: 1000
max_accel: 160000
max_z_velocity: 50
max_z_accel: 1000
max_accel_to_decel: 100000
square_corner_velocity: 18.0

[force_move]
enable_force_move: true

[respond]
default_type: echo
default_prefix:

#####################################################################
#      X/Y/Z Stepper Settings
#####################################################################

[stepper_x]
step_pin: gpio11
dir_pin: gpio10
enable_pin: !gpio12
rotation_distance: 40
microsteps: 16
endstop_pin: tmc2209_stepper_x:virtual_endstop
position_min: -1
position_endstop: -1
position_max: 157
homing_speed: 50
homing_retract_dist: 0

[tmc2209 stepper_x]
uart_pin: gpio9
tx_pin: gpio8
uart_address: 0
run_current: 1.0
stealthchop_threshold: 0
diag_pin: ^gpio4
driver_SGTHRS: 90
interpolate: true

[stepper_y]
[... 565 lines cut ...]
z_hop_speed: 10

[gcode_macro AAA_PROBE_ACCURACY1]
description: CYD_SCREEN_MACRO
gcode:
  G90
  G1 Z20
  G1 X75 Y24 F3000
  PROBE_ACCURACY

[gcode_macro AAC_PROBE_CALIBRATE1]
description: CYD_SCREEN_MACRO
gcode:
  G90
  G0 Z20
  G0 X75 Y24 F3000
  G0 Z5
  PROBE_CALIBRATE SPEED=1000
  Z_OFFSET_APPLY_PROBE

[gcode_macro aab_screws_tilt_calculate1]
description: CYD_SCREEN_MACRO
gcode:
  G28
  SCREWS_TILT_CALCULATE`

JdeV987, 2024-09-07:
This is how mine works:

All of below must be in printer.cfg

#********************************************************************

[output_pin probe_enable]
pin: gpio29                               # The control IO pin 
value: 0                                    # Probe default retracted

# Probe deploy command
[gcode_macro Probe_Deploy] 
gcode:
   SET_PIN PIN=probe_enable VALUE=1

# Probe stow command
[gcode_macro Probe_Stow] 
gcode:
   SET_PIN PIN=probe_enable VALUE=0 

[probe]
# You need to use the z- stop connection for the MicroProbe connection with the black and white wires. 

pin: ^!gpio25                         # The detection IO
deactivate_on_each_sample: False
x_offset: 0.0                         # Actual installed offset of MicroProbe 
y_offset: 0.0                         # Actual installed offset of MicroProbe 
z_offset: 0.0                         # Actual installed offset of MicroProbe 
speed: 10.0

activate_gcode:
    Probe_Deploy
    G4 P500                           # Allow 500 milliseconds for the probe to deploy 
deactivate_gcode:
    Probe_Stow

[stepper_z]
# ? to be replaced by your own values
step_pin: gpio19
dir_pin:  !gpio28                    # You may or may not need to invert the signal with a !  (depending on your Z-motor direction.)
enable_pin: !gpio2
endstop_pin: probe:z_virtual_endstop

microsteps: ?
rotation_distance: ?       
position_min: ?
position_max:  ?
homing_retract_dist: ?
homing_retract_speed: ?
homing_speed: ?
second_homing_speed: ?

#****************************************************
[image]

JdeV987, 2024-09-07:
This is I dot mine to work:
All of below must be in  printer.cfg

`[output_pin probe_enable]
pin: gpio29                                 # The control IO pin 
value: 0                                    # Probe default retracted

# Probe deploy command
[gcode_macro Probe_Deploy] 
gcode:
   SET_PIN PIN=probe_enable VALUE=1

# Probe stow command
[gcode_macro Probe_Stow] 
gcode:
   SET_PIN PIN=probe_enable VALUE=0 

[probe]
# You need to use the z- stop connection for the MicroProbe connection with the black and white wires. 

pin: ^!gpio25                         # The detection IO
deactivate_on_each_sample: False
x_offset: 0.0                         # Actual installed offset of MicroProbe 
y_offset: 0.0                         # Actual installed offset of MicroProbe 
z_offset: 0.0                         # Actual installed offset of MicroProbe 
speed: 10.0

activate_gcode:
    Probe_Deploy
    G4 P500                           # Allow 500 milliseconds for the probe to deploy 
deactivate_gcode:
    Probe_Stow

[stepper_z]
# ? to be replaced by your own values
step_pin: gpio19
dir_pin:  !gpio28                    # You may or may not need to invert the signal with a !  (depending on your Z-motor direction.)
enable_pin: !gpio2
endstop_pin: probe:z_virtual_endstop

microsteps: ?
rotation_distance: ?       
position_min: ?
position_max:  ?
homing_retract_dist: ?
homing_retract_speed: ?
homing_speed: ?
second_homing_speed: ?`

[image]