bigtreetech/MicroProbe #31: Incorrect pins in manual for SKR MINI E3 V3.0?

hrafnfloki, 2024-05-15:
I'm attempting to follow the instructions of the MicroProbe V2 manual, which is on this GitHub, and I've been running into error after error.

>note: this is the location of the previous definition
>  1325 | #define Z_MIN_PROBE_PIN PC13 // The detection IO on the SKR 3 is PC13

>error: 'PE5' was not declared in this scope; did you mean 'PE_5'?

I followed the manual word for word.
So I'm not sure if the manual is out of date, because it was last updated earlier this year? Does anyone know if I'm doing something wrong and what the correct "control and detection I/Os" are?

hrafnfloki, 2024-05-15:
[image]

guiabc38, 2024-10-08:
I have the same problem that you

forplay0709, 2024-11-17:
Yes they are incorrect. I was able to get it to work only if Z_MIN_PROBE_PIN=PC14 and if PROBE_ENABLE_DISABLE=PA1. There's not even a pin described as PE5 on the motherboard. I don't know who wrote the manual and when..