bigtreetech/MicroProbe #33: Problem with firmware - Ender 3 and skr mini e3 v3

AndreiasM5, 2024-07-08:
Hello,

I have an Ender 3 with an SKR Mini E3 V3, and I just bought the Micro Probe V2. I tried both firmware files from the firmware tab, but neither worked.

First, I tried the normal wiring with firmware-ender3-skr-mini-e3-v3.0-microprobev2.bin. When I used this firmware, the printer tried to home and level the bed in the air, not even coming close to the bed. The probe retracts and then doesn’t level again.

Next, I did the other wiring for Z homing and used firmware-ender3-skr-mini-e3-v3.0-microprobev2-for-z-homing.bin. With this firmware, the printer moves to the middle and descends without deploying the probe. The nozzle hits the bed and doesn’t stop unless I switch off the printer.

Can you please help me troubleshoot this issue?

Thank you.

shawnkski, 2024-07-28:
I’m having the exact same issue. It started with the stock board so I switched to the microprobe v2 with no luck then switched to the skr mini e3 v3 and still having the same issues I’ve even switched out the cable for the probes and has made no difference

Egosumumbravir, 2024-08-26:
Sorry for the bad news but those stock firmwares will have been compiled for BL-Touch/CR-Touch probes and their clones.

The v2 BTT microprobes are very much NOT clones of the BL/CR probes and will require custom firmware configurations to work.

guiabc38, 2024-10-07:
Hi :)
Do you solved the isues ?

Egosumumbravir, 2024-10-07:
> Do you solved the isues ?
Yes, the probes needed a custom compile of Marlin to work correctly. I used MRiscoC to get mine working on an Ender v4.2.2 board but never got around to sorting it out for a SKR Mini board before I made the jump straight to Klipper.

One of the nicer features of MRiscoC is the probe deploy/retract buttons and endstop status page so you can see if something is operating as expected.

guiabc38, 2024-10-08:
> One of the nicer features of MRiscoC is the probe deploy/retract buttons and endstop status page so you can see if something is operating as expected.
could you upload the firmware here please?

Egosumumbravir, 2024-10-08:
https://github.com/mriscoc/Ender3V2S1

Musicman3200, 2024-10-29:
Has anyone had any luck with a firmware for a Ender 3 with a SKR mini e3 v3 swap and a Microprobe?   I've tried compiling with multiple versions of Marlin and the only version of the configurations release that has information for the Microprobe is the newest bugfix, yet I can't get it to compile with any version.  I had the same problem to start as @AndreiasM5.  With the same results that can be found in another thread about this same issue.

vitung8, 2024-12-10:
I've been struggling with **_firmware-ender3-bltouch-for-z-homing_** and **_firmware-ender3-bltouch_** for two weeks. Not only does auto-leveling not work in them, the first one doesn't even start printing. In general, the sensor clicks during printing, we cannot control it.
 The only firmware that does auto-leveling is here: https://github.com/bigtreetech/MicroProbe/tree/master/firmware - but they have locked memory! You can't print from them, and there's no source code...
I tried to customize the marlin myself according to the Biqu manual - to no avail. The verdict is this: work through the Klipper or buy another sensor.